Xinyuan-LilyGO/LilyGO-T-SIM7000G #83: At command not responding

4gino, 2021-03-06:
Hi, I' m trying to search for providers using at+cops? But  I end with errorr, and I cannot choose or change provider, It connects to the wrong and weakest, but cannot choose an alternative one. Can you please help. Thanks, i have other simcom modules and menage to control them, but not the one on the lilygo board.

4gino, 2021-03-06:
I mean At+cops=?  ...error

4gino, 2021-03-06:
At+csq give good signal, i tried different sketches, at debug, network test, etc. But when i send at+cops=? The serial monitor stops responding, the only way Is manual reset

WRedux, 2021-03-16:
You've waited?
It took about 15sec. in my Case!
Then it responds with all available Providers...

4gino, 2021-03-16:
Waited minutes

WRedux, 2021-03-16:
Tried CSQ?
Got 15,99

procopio, 2021-03-17:
In the beginning I also tried at+cops=? and the same happened to me. 
I figured out how to live without this command.

WRedux, 2021-03-17:
Have you Guys tried a Modem-Reset?
Could happen, if you mess up with AT-Commands...

LilyGO-T-SIM7000G/examples/Arduino_TinyGSM/ResetModem/

closed 2021-03-28 by 4gino