Xinyuan-LilyGO/LilyGO-T-SIM7000G #85: No IMEI showing after unsuccessful Firmware-Upgrade

WRedux, 2021-03-08:
Hello there,

as in the Title mentioned, I wanted to upgrade my Firmware (03 -> 05).
But my USB-Connection got lost (too quick and dirty USB-Connection).

After fixing this I was able to Upgrade and 05 shows up correctly. 
But in AT+CGSN gives me a Error in SerialMon.
AT+SIMCOMATI leaves the IMEI-Field empty when called.
Also a AT+CSQ gives me 99,99...

Is there any Way to re-flash my Device, so the IMEI would be "written"?

I found this, but that don't concerns my Device:
https://github.com/botletics/SIM7000-LTE-Shield/issues/16

Thanks in Advantage!

lewisxhe, 2021-03-11:
IMEI should not be lost if the upgrade fails. Have you tried to update again?
PS: I have failed several times, but everything works normally after each update. I have not found this problem at present.

lewisxhe, 2021-03-11:
You can try to use this command, write according to the IMEI number on the module, and check again after restarting to see if it is corrected 

[image]

WRedux, 2021-03-12:
Hello,

AT+GSN=? gives OK.
AT+GSN gives ERROR.

found out this:

AT+CPSI?
+CPSI: NO SERVICE,Factory Test Mode
OK

WRedux, 2021-03-12:
I tried several Time to update, even downgrade to 03, which found.
In all Cases Upgrade runs smooth and finished successful, but the IMEI never came back...

WRedux, 2021-03-12:
Solved it :-)
After seeing a ScreenShot of the Flash-Software (SIM7000 QDL V1.43 Only for Update (backup&recover)), I remembered about
the Option to backup/recover the xQCN-File also.... which dissapeared in my Installation of the Software...
So I took a look at the CFG.INI of the Software and found 2 interesting Options, whose set to "false" in my Installation.
1. BacupRestorQcn
2. OnlyQcn

I've set both to true (since the Rest of the Firmware seems OK, only xQCN)...
The missing Options showed after starting the Software and I just reflashed my xQCN.
...and voila:
IMEI shows up correctly.
AT+GSN gives me the correct IMEI, and _AT+SIMCOMATI all correct Data incl. IMEI.
AT+IPSI is now GSM Online... all like before, I guess....

Will go on testing where I left off before Upgrading...and report back...

The Hint with the crashed xQCN-File I got from the [Botletics-Guys](https://github.com/botletics/SIM7000-LTE-Shield/issues/16)

closed 2021-03-16 by WRedux