Xinyuan-LilyGO/LilyGO-T-SIM7000G #87: Core panic Illegal Instruction Error on start up.

TravisDuck999, 2021-03-13:
Hi, I have been hitting problems that I did not expect or I can find anywhere else online with this board, I am hoping you can help.
I just got 2 new LilyGo SIM7000G's the newer 2020 version.

The first one checked my SIM card, plugged in the anetnna's and put in the SIM card and hit upload.
I have constantly been getting Illegal Instruction errors.

I think it has something to do with the Serial connection.
I have tried a basic blink sketch to make sure the ESP32 works, that is fine.

The full error the ESP32 spits out on boot is :
```
[6859] Trying baud rate 115200 ...
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400d0e3c: 0000002d ffffffff ffffffff
Core 1 register dump:
PC      : 0x400d0e41  PS      : 0x00060530  A0      : 0x800d0f77  A1      : 0x3ffb1f20  
A2      : 0x3ffbff24  A3      : 0x3ffbff40  A4      : 0x0001c200  A5      : 0x00000000  
A6      : 0x3f400122  A7      : 0x0000000a  A8      : 0x800d0e2c  A9      : 0x3ffb1f00  
A10     : 0x0000000a  A11     : 0x0000000a  A12     : 0x0800001c  A13     : 0x00000009  
A14     : 0x00000009  A15     : 0x00000000  SAR     : 0x00000016  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400882d5  LEND    : 0x400882e5  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d0e41:0x3ffb1f20 0x400d0f74:0x3ffb1f80 0x400d1939:0x3ffb1fb0 0x40089c09:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
```

This is with the AT_Debug sketch in the Tiny GSM library.
It seems to just not corrently talk to the SIM7000G no matter what I do. ( I have tried hardcoding the baud rate )

I battled with this for 2 days, trying bigger power supplies and lots of other sketchs and libraries but no luck.

I then plugged in the other board thinking the first one was broken.
I got AT command outputs almost immeditaly without uploading anything ( picutres attached ) 
WITHOUT the GPS antenna attached, with the antenna attached I dont think my laptop USB port can supply the current to power everything.

I am perfectly happy not having GPS functionality for this porject if it matters.

The thing that has completly thrown me is now when I upload the AT_debug sketch I get the SAME Illegal Instruction error.

Is there any chance I could see the code that gets loaded on the board before it is shipped ?
Maybe I can get it working from that, or at least upload it and see if it works again as it did.

The ony other thing I can think of is that maybe I'm selecting the wrong upload options on Arduino IDE ?
I have tried the ESP Wrover module and the ESP32 Dev module.

The output before I uploaded code to the second LilyGo SIM7000G is below.

With SIM card and antenna attached.
[image]

Without Sim card.

[image]

Any ideas or pointers would be fantastic as I can find no one else that has this problem online, if you have access to the production code of the LilyGo T-SIM7000G that I could look over that would be amazing, because something is not right here.

WRedux, 2021-03-15:
Hello,

Please try 
LilyGO-T-SIM7000G/examples/Arduino_TinyGSM/ATdebug/
instead of AT_Debug, got the the same Guru-Error with this.
Works for me...

TravisDuck999, 2021-03-16:
Hi, yea I think that was my problem, weird but in the end of the day it works now and I dont really need the TinyGSM library as I'm planning to do everything with AT commands anyway.

For anyone else getting this problem, only use the examples on the SIM7000 Git page as stated above, the bundeled examples with TinyGSM did not work for me at all.

Thanks for the help.

closed 2021-03-16 by TravisDuck999