Xinyuan-LilyGO/LilyGO-T-SIM7000G #90: Platformio example USB Battery data transfer

4gino, 2021-03-21:
I'm using the platformio example code, but I' m getting some strange results. When USB Is connected and serial monitor open Data transmission goes well, modem on , signal quality good, data sent, modem off. When I disconnect the USB and power the board with a battery (new and charged to 4.2v) data Is not sent. Probably the signal quality Is worse, or something else but i cannot see It with the serial monitor, because if I connect the USB everything will go well again. Anyone else had the same issue? The Sim7000 receive different type of Power from USB or from battery?  Thanks

ejri, 2021-03-22:
Yeah, if you check the readme page, there are some issues with the same pins being used for more than one thing at a time. So, you have to make sure to check which pins are being used depending on the connected power source...

4gino, 2021-03-23:
Solved, was software related. When used with USB data sent for battery status Is 0. When battery powered the number Is different from 0 and there was a bad code in my new lines. The point Is that i could use serial monitor connecting just tx rx gnd pins, so the battery was the power source, and serial monitor gave me the correct debug.

closed 2021-03-25 by 4gino