Xinyuan-LilyGO/LilyGO-T-SIM7000G #91: Multiple card GPS stops working without explanation

eueduardo3, 2021-03-24:
GPS simply stops working without explanation, it was days running. More than one plate, which were in different locations, had the same defect

lewisxhe, 2021-03-29:
Have you used this test? 

https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/tree/master/examples/Arduino_GPSTest

Mr-HaleYa, 2021-03-30:
you really didn't explain it that well...
code sample? location? sim carrier?

eueduardo3, 2021-03-30:
> you really didn't explain it that well...
> code sample? location? sim carrier?

I have some T-SIM7000 boards. With any script that uses GPS, none of them restore the location.

The modem network celullar functions are ok.

Mr-HaleYa, 2021-04-12:
It can take up to 3 minutes for it to find a GPS location and you should be outside in an open area or it will fail. AT+CGNSINF just means its searching

code I use for GPS, works perfect on all 40 of our boards
```cpp
void getGPS() {
  float lat=0;
  float lon=0;
  Serial.println("Turning on GPS  power");
  modem.sendAT("+SGPIO=0,4,1,1"); // V20200415 set GPIO4 HIGH to turn on GPS power
  delay(1000);
  modem.enableGPS();
  long startTime = millis();
  Serial.println("Scanning for location for 2 mins");
  while (millis() - startTime <= 120000 ) {    // scans for 1 min 
    if (modem.getGPS(&lat, &lon)) {
      Serial.printf("lat:%f lon:%f\n", lat, lon);
      break;
    }
    delay(2000);
  }
  if (lat == 0) { Serial.println("Failed to get location"); }
  modem.disableGPS();
  Serial.println("Turning off GPS power");
  delay(1000);
  modem.sendAT("+SGPIO=0,4,1,0"); // V20200415 set GPIO4 LOW to turn off GPS power
}
```

CyprienHoelzl, 2021-04-20:
I have a similar issue with my first ever unit. Everything works, except GPS. Even after waiting hours, it doesn't fix the location.

eueduardo3, 2021-04-25:
> It can take up to 3 minutes for it to find a GPS location and you should be outside in an open area or it will fail. AT+CGNSINF just means its searching
> 
> code I use for GPS, works perfect on all 40 of our boards
> 
> ```c++
> void getGPS() {
>   float lat=0;
>   float lon=0;
>   Serial.println("Turning on GPS  power");
**>   modem.sendAT("+SGPIO=0,4,1,1"); // V20200415 set GPIO4 HIGH to turn on GPS power
> delay(1000);
>   modem.enableGPS();**
>   long startTime = millis();
>   Serial.println("Scanning for location for 2 mins");
>   while (millis() - startTime <= 120000 ) {    // scans for 1 min
>     if [...]

Thanks for posting, I recovered some of the devices by placing the 1 second delay you used between modem.sendAT ("+ SGPIO = 0,4,1,1") and modem.enableGPS ()

Mr-HaleYa, 2021-04-25:
@eueduardo3 why did you reopen it then???

eueduardo3, 2021-04-28:
> @eueduardo3 why did you reopen this then???

eueduardo3, 2021-04-28:
> @eueduardo3 why did you reopen this then???

so that this answer can help other people who have a problem with GPS also because of this delay. I hadn’t seen anywhere commenting that this delay is necessary to avoid GPS problems

closed 2021-04-28 by eueduardo3