Xinyuan-LilyGO/LilyGO-T-SIM7000G #93: Solar Max voltage Vs Open circuit voltage ?

TravisDuck999, 2021-03-26:
I see a lot of people on here using a 6V solar panle and having good results.

I am looking at buying a solar panel for my unit and am wondering if the Voc (Voltage open circuit) makes a difference?

I see many suppliers selling 6V panels but the Voc is around 7V will this mess with the solar input (4.4 - 6v for the CN3065 ) ?

Has anyone tried measuring their 6V solar panel when disconnected under full sun to see if it goes above when there is no load ? ( and has confirmation that it works for at least a few months with the SIM7000G ) 

My concern is that the open circuit voltage will break something when the battery is full charged and nothing is needed from the panel as such the voltage will rise to Voc and break the CN3065 or something else.

Hope this makes sense.

Mr-HaleYa, 2021-03-30:
ya I get that but don't worry about it too much, I have used many different panels and most have had Voc higher than 6, what I have looked for is the Vmp to be ~6 and not high because that's the max voltage it will hit when being used

I understand the fear tho, like my 12v panels are
Voltage MPP Vmp(V): 17.80V
Voltage Open Circuit Voc(V): 21.15V

these work great https://www.amazon.com/ALLPOWERS-Battery-Charger-Encapsulated-130x150mm/dp/B074TYH68Z/ref=sr_1_3?crid=9NYT45S7J9P&dchild=1&keywords=5v+solar+panel&qid=1617083762&s=lawn-garden&sprefix=5v+sol%2Clawngarden%2C242&sr=1-3

If this answered your question please close

TravisDuck999, 2021-04-01:
Thanks for the info, really freaking scary to try it as I'm sure you understand.

I am looking at your link and it seems to be a Voc of 5.91V so safe with the 6V limit.
But a little higher would yeild better results.

If anyone out there could try an open circuit voltage test with one of their 6V panels it would set my mind at ease and I'm sure many other people.

If not I will close this issue in a week from today.

Mr-HaleYa, 2021-04-01:
it won't hit the open-circuit voltage even if the batteries are full because reasons 😄 

If you want to be really safe you could do what I do and take the pack off the back and solder on a larger one (MAKE SURE THEY ARE IN PARALLEL!!! so the total voltage of the pack is still 2.5-4.2 volts 3.7 nominal) then I use a solar charge chip and 12v 10w panels. boosts battery life and charges fast

[solar panels](https://www.amazon.com/dp/B00W80N8TA/?coliid=I1CY9DAGNWO3BC&colid=20Q85TRFN2F5A&psc=1) form amazon 12v 10w

[Charge chip](https://www.aliexpress.com/item/1005001962453608.html?spm=a2g0o.productlist.0.0.41735e8fjLkIEm&algo_pvid=3ab5b8ef-3b3d-4d06-817e-1b6da0388107&algo_expid=3ab5b8ef-3b3d-4d06-817e-1b6da0388107-9&btsid=0b0a555316173178430425510ed5de&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_) From aliExpress but can be found on Amazon too, just make sure its 1s output (made to charge 3.7v lithium battery NOT 12v lead acid)

TravisDuck999, 2021-04-07:
Thanks for the help, I am going to try with a 6V normal solar panel and see what happens, I will reopen this is it breaks to warn anyone else. If this is still closed in about a month then the 6V solar panel normal operation works fine for me.

closed 2021-04-07 by TravisDuck999

gasagna, 2022-10-23:
Hi @TravisDuck999, you said you tried with a 6V normal solar panel, but did you mean a panel with Voc > 6V? What solar panel are you using? Thanks!

TravisDuck999, 2022-10-23:
Hey @gasagna so the panel I bought didn't give me those specs, it was this one though : 
https://www.robotics.org.za/6V35W

Unfortunately I don't have the panel with me and won't for at least a few months or I would meter it out for you.