Xinyuan-LilyGO/LilyGO-T-SIM7000G #94: Li-ion battery pack connection doubt

ishdemon, 2021-03-26:
where should i connect an external battery (Not 18650)? should i solder it to the + & - of the battery holder and will it get charged normally also?

ishdemon, 2021-03-26:
also charging is only done when USB is connected or it will also charge when i supply power via 5v & GND Pins??

lewisxhe, 2021-03-29:
That’s right. You can only solder + and-on the back. You can’t connect it directly to 5V. It’s a voltage of 5V. If you connect it directly, it will be damaged.

ishdemon, 2021-03-29:
no i meant external power supply on 5v & GND pins. Will this charge the battery connected.?

lewisxhe, 2021-03-29:
@ishdemon Yes,it can be chargeing

Mr-HaleYa, 2021-03-30:
@lewisxhe @ishdemon Your question was answered. close this or ask another one

closed 2021-03-30 by ishdemon