Xinyuan-LilyGO/LilyGO-T-SIM7000G #96: Bluetooth temperature sensors

4gino, 2021-03-28:
Hi, anyone here menaged to connect a Bluetooth temperature humidity sensor?  Is there any example code? or do you know how much current will the lilygo board consume in an open field application where multiple sensors send data to the board? Thanks a lot

lewisxhe, 2021-03-29:
There are simple examples of Bluetooth usage in esp32 core, and the usage of Bluetooth has nothing to do with the board.

Mr-HaleYa, 2021-03-30:
@4gino ya this isn't the place to be asking that question, go to Arduino forum for help with basic sensor stuff 
@lewisxhe this should be closed....

4gino, 2021-03-31:
I'm Sorry, probably i did not explain well. I see in other forum that using Bluetooth with esp32 Is not Easy, working at 240 MHz and in lightsleep the Power consumption Will be too much, anyone here menaged to lower It, or use Bluetooth with battery Power?

Mr-HaleYa, 2021-04-01:
Of course, you can use Bluetooth with a battery... It might not last that long.
what is the point of this issue???

> anyone here managed to connect a Bluetooth temperature humidity sensor

yes, it's not that hard.

>Is there any example code?

google "Bluetooth sensor esp32"

>do you know how much current will the lilygo board consume in an open field application where multiple sensors send data to the board? 

no, you have to do your own testing because every sensor uses a different amount of current

closed 2021-07-12 by 4gino