Xinyuan-LilyGO/LilyGO-T-SIM7000G #97: dying boards...

Mr-HaleYa, 2021-03-30:
I have had 3 boards die ( technically 4 but that was my bad, connected battery backward and poof it dead ) 
I am using an external battery back 3s3p 18650s that output ~11.1v, this is inputted into a 7 - 36 VDC to 5v 1A step-down transformer TSR 1-2450E ([tsr1e_datasheet.pdf](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/files/6226540/tsr1e_datasheet.pdf)) measuring the output I get 5.01v so this 5v power is connected to the Vin pin on the TTGO board and is what I have been using to power the 3 boards that have died... the boards work fine for a while then just die. 

Two have stopped communicating with my PC so I think the CP2102  host bridge chip is dead... They still work just fine other than that as far as I can tell. The ESP32 still runs my code and the Sim7000G still works because it still sends me data, however it is very annoying that I can't upload new code using the USB interface.

The 3rd board's modem will not stay powered on, so I flashed a script the outputs the modems responses into the serial monitor and I get OVER-VOLTAGE POWER DOWN meaning it's getting too much voltage on the Vbat pin of the Sim7000G which are pins 55,56,57 [SIM7000 Hardware Design_V1.05.pdf](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/files/6226807/SIM7000.Hardware.Design_V1.05.pdf) they have a voltage range of 3.0v to 4.3v, anything higher or lower and it will shut down the modem. when I measure a working board I get ~3.8v but when I measure the damaged one I am getting 4.6v! @lewisxhe What is the Power IC for the Sim7000G and where on the PCB is it so I can see if I can replace it or look at the datasheet.

I don't understand what is happening... Vin is a 5v input and I am inputting exactly 5v, when I measure Vin with USB connected it is 4.8v so is it NOT a 5v in??? this is super confusing and I just want to know why they are dying when everything says they shouldn't be. Basically, it seems that some kind of power IC is failing and it is causing other parts of the board to die even tho i am using correct voltages.

ishdemon, 2021-03-30:
damn, so the only safe way to power these boards is via USB C. is it?? @lewisxhe

lewisxhe, 2021-03-30:
TYPE-C and Vin are connected together, so you can input 5V through Vin, you can view the schematic 

@Mr-HaleYa

lewisxhe, 2021-03-30:
Can't communicate with PC now, can I see Cp210x in device manager?

Mr-HaleYa, 2021-03-30:
> Can't communicate with PC now, can I see Cp210x in device manager? 

No, it does not connect to the pc it does not show up in device manager and it does not make the connection sound. The board does get power from the usb tho

A working board does show silicon labs Cp210x

Mr-HaleYa, 2021-03-30:
> damn, so the only safe way to power these boards is via USB C. is it?? @lewisxhe

I have 14 running on battery packs and none of them have died. Just the ones using the Vin pin have died (which I think it might be more of a coincidence then causation as of now)

ishdemon, 2021-03-30:
> I have 14 running on battery packs and none of them have died. Just the ones using the Vin pin have died (which I think it might be more of a coincidence then causation as of now)
I was planning to use this board in a car with a buck converter to convert 12V to 5V on VIN and GND.  i guess voltage greater than 5V on VIN pins might damage the board in-case of voltage fluctuations from the converter. What is the source of your supply??

ishdemon, 2021-03-30:
> > damn, so the only safe way to power these boards is via USB C. is it?? @lewisxhe
> 
> I have 14 running on battery packs and none of them have died. Just the ones using the Vin pin have died (which I think it might be more of a coincidence then causation as of now)

I was planning to use this board in a car with a buck converter to convert 12V to 5V on VIN and GND.  i guess voltage greater than 5V on VIN pins might damage the board in-case of voltage fluctuations from the converter. What is the source of your supply??

lewisxhe, 2021-03-30:
@Mr-HaleYa You can check the schematic, Vin and TYPE-C VBUS are connected together, it shouldn’t happen, I suspect it is caused by other problems

Mr-HaleYa, 2021-03-30:
> What is the source of your supply?? @Mr-HaleYa

@ishdemon TSR 1-2450E . They're very efficient and very small like about the size of a tic tac, cost about $5 each and they are just really good. They can deliver 1 amp continuous just fine and I've done voltage testing and I've never had voltage spikes or the voltage raise above 5.02 volts unless I exceed the maximum input voltage of 36 v then it spikes right before it kills the transformer but like I said in the first post I'm supplying ~11.1v into the step down transformer so well within its range.

You can just Google what I put or put a link to the datasheet in the OG post

Mr-HaleYa, 2021-03-31:
> I have a similar setup to yours except for safety I put a L7805 LDO voltage regulator with the recommended capacitors between the external MPPT and the VBAT pin

so like the exact same thing just a different regulator... And you have not had any die? how many do you have like this? How long? And what is the voltage going into the regulator?

rnborland, 2021-04-01:
I'm developing a solution based on this panel https://www.amazon.com/gp/product/B07ZGNC4HD/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1
I want to use it because of the weather protection and mounting hardware is god, it outputs 6 volt pulses, It has a feature where it wont charge unless it senses residual voltage from the battery so i run a sense line to the positive terminal of the battery through a diode. I use a voltage regulator to limit the voltage going to the battery. Im still tweaking the software etc to correct some issues e.g going in to deep sleep if the battery voltage is low after a couple of "no sunshine" days rather than run the battery down to where it wont charge. Some other "quirks' are that If you have any sensors on GPIO2 you cant upload code. I had a DHT 11 on pin GPIO2 and it caused this problem.
So far I've built one system for demo purposes and still testing/improving before I deploy to customers. I have a custom interface board that I made with i2C grove sockets that works nice and have tested a number of sensors with it. The idea is to be able to offer prototype systems quickly with cloud uploads and web access to the data.  I have also been experimenting with upgraded antennas . I have more boards and processors on order .
[image]

Mr-HaleYa, 2021-04-01:
@rnborland Thank you for the comment but that doesn't really help me... 

So these are my boards and they have many different ways we can configure them, just for reference the only ones having the problem are the ones being powered via the Vin pin (5v)
[image]

We build our products for irrigation management companies so they can track water usage through our state. I recently decided to add an optional camera upgrade to allow for remote viewing of a site. the only problem is that the camera is 5v only so we either needed to add a 3.3 to 5v boost or just power the whole board off of 5v, eventually, we decided on the latter. 

Here is one in a different config just so you can see 
[image]

Like I said the boards are multi-purpose so for one setup we use it to drive a very large stepper motor that takes 24v and in another case, we have it just running the camera, so we have 2 different battery types we use when doing this

[image]
this is a 3s3p ~11.1v battery pack for the camera stations (that's what we call them when they are all built) the pack has a balance charger and we charge them with 10w 12v panels.

[image]
And this is a 2s2p 24v battery pack for stations that require a stepper motor and we charge the batteries with a dedicated charger that uses a 30w 24v solar panel.
 
So that is why we use the step down that has an input voltage range of 7-36v 

In both cases the voltage is put through the step-down converter so that it is 5v then it is used to power the boards. 

---

Sooooo ya I don't understand why the boards are dying... as you can see in the first image I have removed the single-cell battery pack on the back of the ttgo and replaced it with a 4 cell pack to allow for longer life (it still connects to the + pin on the back of the board and GND) I have never had a board die that uses the battery connected to the ttgo (basically ones powered with 3.7v) 

I did notice a weird thing before all of the boards have died, they start giving a very high voltage reading on the Vbat pin, exactly 7.2v on all of them... I remove the batteries that connect to the ttgo when I'm using the 5v pin so it's not constantly trying to charge them for no reason. (Side note: I have left them in before and a board still died so removing/keeping them is not the cause) anyways, I measure the Vbat pin to keep track of the onboard battery voltage and when there are no batteries connect and I'm just using the Vin pin the board Vbat is obviously 0v but right before they die it spikes and holds at that 7.2v for minutes... hours... I've had both occur, then it dies... 

you can see here on my Vbat voltage graph that it's normally 0 but then spikes to 7.2 then goes down a bit then spikes again and dies at 6:40 PM on my desk right in front of me... I didn't touch it, I didn't do anything, just was chilling then noticed it wasn't working anymore...
[image]

Here is an image of the 11.1v battery pack at the same time (this is what was being used to power the board) 
It was normally at 12.59v (fully charged) then it spiked to 16.22v. This can't be true because I had a voltage meter connected to it the whole time and it logs its voltage and it NEVER went up it was actually slowly discharging like a battery is supposed too
[image]

I measure the external battery packs (11.1v and 24v) with GPIO35 and the internal batteries (4 on the board 3.7v) with GPIO34 because that's the one connected to Vbat by default. 
When measuring the 11.1v battery pack I have a voltage divider of 43kΩ and 10kΩ so the voltage that goes to the pin is a max of 2.377v when the pack is fully charged at12.6v and 1.698v when at 9v (this is auto cutoff voltage)
When measuring the 24v battery pack I have a voltage divider of 100kΩ and 10kΩ so the voltage that goes to the pin is a max of 3v when the pack is charging at 33v and 2v when at 22v (this is auto cutoff voltage)
[image]

@lewisxhe this is just a spitball but do you think having the boards off but the voltage still going through the voltage divider and into the pins is killing the boards??? It only ~1.5v so defiantly not enough to back-power the board through the pins but idk maybe enough to kill parts??? only the 11.1v and 24v stations have died ant there the only ones with the voltage divider that goes to GPIO35

so ya idk what to do, something is killing them and it kills the pins I use to read the batteries to 
here's a text file of what's happening (the symptoms) of the two dead boards I currently have on hand
[Dead boards.txt](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/files/6246236/Dead.boards.txt)

any suggestions? or questions? I REALLY need to figure this out because I am supposed to be installed like 20 more stations next month and 5 of which need cameras but I don't want them dying when a customer is using them.

rnborland, 2021-04-02:
Mr Hale, Sorry my last post wasn't much help. Based on your more detailed problem description I did have an issue a year or so ago where I was using regular esp 32 wifi boards to switch a 12 volt solenoid valve using an n-channel MOSFET. I damaged a few boards while troubleshooting the problem. The issue was that while switching the valve off and on there was very high voltage and very short duration voltage spikes going back though my board burning out the USB to serial chip and also the on board voltage regulator. The board would stop working and get hot ( like yours)  

The solution was to incorporate large enough capacitors in both sides of my external voltage regulator circuit and diodes to help prevent back EMF coming back through my +ve line. Maybe you have a similar issue when switching on/off your camera and/or stepper motor circuit maybe not but its worth checking out.

Mr-HaleYa, 2021-04-10:
That actually sounds possible. I was thinking it was voltage spikes but I don't have any proof, I don't know how to capture fast voltage spikes since I honestly just started learning oscilloscopes at the college but maybe I can get a professor to show me how.

Anyways @rnborland what size caps did you use??? I currently use [this](https://lcsc.com/product-detail/Aluminum-Electrolytic-Capacitors-Leaded_CX-Dongguan-Chengxing-Elec-C43329_C43329.html) for my 3.3v output that connects to my low voltage sensors. It is a 100uF 10V electrolytic capacitor and works great for smoothing voltage dips when I turn on sensors. I was thinking of using a 50v 100uF electrolytic capacitor like [this](https://lcsc.com/product-detail/Aluminum-Electrolytic-Capacitors-Leaded_CX-Dongguan-Chengxing-Elec-100uF-50V-Black-high-frequency_C47891.html) for the power going into the 5v stepdown transformer. Is 100uf sufficient or should I go higher or lower?

Mr-HaleYa, 2021-05-06:
I went and got a Rigol DS1054Z oscope and have been measuring the voltage spikes when different current draws are applied by different sensors turning on and off and I think I see now why they were dying.
[image]
(10x probe)
these images are from esp startup and you can see that there are voltage spikes OVER 9V!!! 
5v (esp disconnected) -> 9v (instant esp connected) -> 5v
I think the super quick spikes might be fine when the esp is going off to on but I think it's becoming dangerous when the camera is switching on and off and causing spike while the esp is on so then its blowing parts bc all the startup caps are saturated and nothing can absorb the voltage spike. It's not so extreme tho since the camera has some onboard caps so it can smooth most of its voltage however it's not perfect. 

___

[image]
(1x probe)
when esp is connected to power with no caps it gets a spike of 8.28V
[image]
(1x probe)
when esp+cam get connected to power at the same time gets a smaller spike of 5.9V due to the camera having internal caps 
[image]
(1x probe)
The camera switching on with esp already connected makes a very small voltage spike of 5.6V
[image]
(1x probe)
esp+cam+10v 200µf cap gives this super smooth startup

___

The camera has a startup current draw of ~300mA whereas the TTGO plus all the connected sensors have a startup current of ~1.4A if I did my math right, so I am also upgrading the TSR to the 1.5A version instead of the 1A version

I'm fairly confident in this being the solution so I will order some new PCBs and run more tests
🤞 hope this was the problem
🙏 pray this was the ONLY problem 😆

rnborland, 2021-05-06:
Looks good, would definitely head in the direction of having way more capacitance than you need across the load and backing off based on tests. Also, you might want to look at diodes.

Mr-HaleYa, 2021-05-06:
I'm going to use [these](https://lcsc.com/product-detail/Aluminum-Electrolytic-Capacitors-Leaded_CX-Dongguan-Chengxing-Elec-C45081_C45081.html) capacitors that are rated 35V 470uF. Planning on placing them on both sides of the 5v transformer as well as the 3.3v line to stabilize my sensor's power supply when they are turning on and off. 

My only problems with diodes are the voltage drop and I have no reverse polarity voltage 
If I had a motor or something that can create a reversed polarity I would defiantly add a flyback diode. the stepper motor has its own driver board and it's completely isolated from the rest of the components on the PCB

rnborland, 2022-01-07:
Lilygo I have seen some brownout issues which happen when the battery voltage gets less than 4 volts or so. What happens is the board wakes up from deep sleep, then brown out-reboot-brown out ..etc until the battery is drained. I am trying a capacitor between 3.3v and gnd as suggested by Travis on an earlier closed post to see if it helps. I am using the All Functions sketch with a BME280 sensor.

DomolinBolivia, 2022-02-22:
I am using SOLAR_IN with (4.4v-6v), with a 7805 voltage regulator as the 12 VDC input of a vehicle, first day without any problem yet, could I have a problem in the future?

hardastyle, 2022-02-24:
Were you guys able to come up with the reason why they die and what change completely reaolves the issue?

DaviesAguirreBlas, 2024-01-16:
> I am using SOLAR_IN with (4.4v-6v), with a 7805 voltage regulator as the 12 VDC input of a vehicle, first day without any problem yet, could I have a problem in the future?

Hey @DomolinBolivia ,

Any updates of your test?