Xinyuan-LilyGO/LilyGO-T-SIM7000G #98: ThingsBoard Example gets stuck in GPS loop

ishdemon, 2021-03-30:
The sketch gets stuck in `modem.getGps()`.`AT+CGNSINF`, just doesn't reply at all, stuck in a loop :

AT+CGNSINF
AT+CGNSINF
AT+CGNSINF
AT+CGNSINF
AT+CGNSINF
AT+CGNSINF
AT+CGNSINF

Sometimes it works after reuploading the sketch or hard reset button. Also happens with the GPSTest sketch too.

ishdemon, 2021-03-30:
So far i have found similar issue happening in another board https://github.com/botletics/SIM7000-LTE-Shield/issues/57.

Mr-HaleYa, 2021-04-01:
It can take up to 3 minutes for it to find a GPS location and you should be outside in an open area or it will fail. AT+CGNSINF just means its searching

code I use for GPS, works perfect on all 40 of our boards
```cpp
void getGPS() {
  float lat=0;
  float lon=0;
  Serial.println("Turning on GPS  power");
  modem.sendAT("+SGPIO=0,4,1,1"); // V20200415 set GPIO4 HIGH to turn on GPS power
  delay(1000);
  modem.enableGPS();
  long startTime = millis();
  Serial.println("Scanning for location for 2 mins");
  while (millis() - startTime <= 120000 ) {    // scans for 1 min 
    if (modem.getGPS(&lat, &lon)) {
      Serial.printf("lat:%f lon:%f\n", lat, lon);
      break;
    }
    delay(2000);
  }
  if (lat == 0) { Serial.println("Failed to get location"); }
  modem.disableGPS();
  Serial.println("Turning off GPS power");
  delay(1000);
  modem.sendAT("+SGPIO=0,4,1,0"); // V20200415 set GPIO4 LOW to turn off GPS power
}
```

ishdemon, 2021-04-11:
I see there is a delay between :
```
modem.sendAT("+SGPIO=0,4,1,1"); // V20200415 set GPIO4 HIGH to turn on GPS power
delay(1000);
modem.enableGPS();
```
is it significant?

Mr-HaleYa, 2021-04-11:
It gives it time to turn on the power to the GPS antenna, remove it if you feel like it, otherwise leave it alone.

ishdemon, 2021-04-12:
@Mr-HaleYa Also when should we do Modem Hardware reset, in the example code Modem reset is done when it fails to get GPS lock,fails to connect to the server etc. So is there any criteria for Modem Reset?

ishdemon, 2021-04-12:
> It can take up to 3 minutes for it to find a GPS location and you should be outside in an open area or it will fail. AT+CGNSINF just means its searching
> 
> code I use for GPS, works perfect on all 40 of our boards
> 
> ```c++
> void getGPS() {
>   float lat=0;
>   float lon=0;
>   Serial.println("Turning on GPS  power");
>   modem.sendAT("+SGPIO=0,4,1,1"); // V20200415 set GPIO4 HIGH to turn on GPS power
>   delay(1000);
>   modem.enableGPS();
>   long startTime = millis();
>   Serial.println("Scanning for location for 2 mins");
>   while (millis() - startTime <= 120000 ) {    // scans for 1 min 
>     if (modem.getGPS(&lat, &lon)) {
>       Serial.printf("lat:%f lon:%f\n", lat, lon);
>       break;
>     }
>     delay(2000);
>   }
>   if (lat == 0) { Serial.println("Failed to get location"); }
>   modem.disableGPS();
>   Serial.println("Turning off GPS power");
>   delay(1000);
>   modem.sendAT("+SGPIO=0,4,1,0"); // V20200415 set GPIO4 LOW to turn off GPS power
> }
> ```

tried this code, been an hour, not able to get lock. But when i try the ThingsBoard example ..it gets locked eventually and also this reply `+CGNSINF: 1,0,,,,,,,0,,,,,,29,,,,41,,` comes few times and it gets locked after few tries. Code seems exactly same apart from few delays..no idea why it works there but not in my sketch. Is there some kind of proper boot initialize sequence that we need to follow?

ishdemon, 2021-04-12:
Update : Not able to get GPS lock with any code now. It seems something wrong with the board? Do i need to updated the SIMCOM firmware? i have the newer board 20200415

closed 2021-04-13 by ishdemon