Xinyuan-LilyGO/LilyGO-T-SIM7000G #99: Importing GPS Library In Arduino IDE

Memelord-The-First, 2021-03-31:
So i'm relatively new to Arduino and I was wondering how I can import this Library into the IDE. Unlike other packages, I can't just download the ZIP File and import it that way because it doesn't work. I tried just importing the "firmware" Folder and that doesn't work either. I only want to be able to use the GPS Capabilities of the board as everything else is working with the Wrover B Library. Any help would be appreciated!

PS: Also it would be helpful if someone could tell me the Functions used to get the Latitude and Longitude.

ishdemon, 2021-04-01:
You need TinyGSM library to use the GPS capabilities. Import that library in Arduino/PlatformIo.

Mr-HaleYa, 2021-04-01:
@Memelord-The-First are you using Arduino IDE or platform IO ?

Memelord-The-First, 2021-04-02:
I'm using Arduino. Never heard of Platform IO.

Mr-HaleYa, 2021-04-02:
good I strongly dislike platform IO, it doesn't work for like 80% of the jobs I do 😅 

so you need to install the TinyGSM library and you can follow the instructions here to do that https://www.arduino.cc/en/guide/libraries

then you can use the examples from the TinyGSM library https://www.arduino.cc/en/Tutorial/LibraryExamples

or you can download the examples from this repo https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/tree/master/examples
open the one you want, open the file, then click the download button

Memelord-The-First, 2021-04-02:
Thanks for such a detailed answer! Will hopefully try it tomorrow and close up this issue

Mr-HaleYa, 2021-04-12:
Did it work? If so, close this