Xinyuan-LilyGO/LilyGO-T-SIM7000G #101: Information about Battery

carminelau, 2021-04-03:
Hi, I would like to use the 18650 battery but at the same time power the esp32 with the type c.

When the power fails, does the 18650 battery automatically go into operation?

Thanks and happy easter

TravisDuck999, 2021-04-04:
I am 99% sure that is how it works.
As long as your switch is on the ON position it should take over automatically.

It is a battery backup after all.

carminelau, 2021-04-04:
thanks for the reply, you were very kind.
I inserted the battery in the compartment underneath the ESP32 but when there is a power failure in the type c the ESP 32 remains on, but cannot communicate with the simcom modem

TravisDuck999, 2021-04-04:
Hmmm, ok well to be honest I have just started playing with this board, maybe this is a question better answered by someone with more experince..

FZELECTRONICS, 2021-04-07:
yes! that's exactly what happens! but you may experience a restart, at least if you pull the USB C out, sometimes the esp32 restarts even with the battery fully charged, just to keep in mind

Mr-HaleYa, 2021-05-06:
@carminelau Your question was answered, close this

fgnievinski, 2023-06-06:
here it says you need to insert a capacitor to avoid restarting during power hand-over from USB-C to battery:
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/123#issuecomment-874655793