Xinyuan-LilyGO/LilyGO-T-SIM7000G #102: IO0 FUNCTION

FZELECTRONICS, 2021-04-06:
Hi, could anyone tell me which is the specific use of the IO0 pin?

I know it has something to do with DTR and RTS pins but can't get the specific functionality.
could I use it as a digital output without breaking the UART interface? my guess is that I can but just to make sure.

Thanks and regards!
great dev board by the way!

lewisxhe, 2021-04-08:
GPIO0 is used as the startup setting. If you use it as an output, you can use it, provided that the output load cannot have pull-up or pull-down attributes, otherwise it will affect the download and normal startup functions

Mr-HaleYa, 2021-05-06:
@FZELECTRONICS your question was answered close this

closed 2021-10-19 by LilyGO