Xinyuan-LilyGO/LilyGO-T-SIM7000G #103: Tracker doesn't work when switching from USB power to 18650 battery power

loctracker, 2021-04-07:
I worked on some code for a MQTT tracker. It works fine but when I switch from USB power to 18650 battery power, it doesn't seem to work. To be clear, I turn the power on with the switch with only the battery connected. Without USB I loose the ability to see what is happing in the terminal window. No data is send to io.adafruit.com. Is it stuck at trying to communicate with the USB port?
It seems to be doing something, the current from the battery varies between 50 and 200mA. 

Any help is greatly appreciated!

FZELECTRONICS, 2021-04-07:
hi, why dont you connect gnd, tx0 and rx0 with a serial to USB interface to see what is happening in the terminal without plugin in the USB C of the board.
(UART0 outputs same as USB C port)
hope it helps

loctracker, 2021-04-07:
Thanks, good idea! I am thinking of inserting the 18650 battery and connecting the USB-c cable while disconnecting the red power wire in the USB-c cable. I made the cable for measuring current though USB. That way I don't have to solder connections to the board.

loctracker, 2021-04-07:
Well that didn't work, I assume that the USB power line powers some serial logic on the board. But when I connected the battery and only the battery on the board it seems to work fine, data was sent to io.adafruit.com. Before, when it failed to sent data I tried to measure the current from the battery and now I am assuming that the internal resistance of the meter was too high and the ESP32 crashed or froze. I was using a digital multimeter on 200mA DC.

dudemyass, 2021-04-10:
@loctracker 
can you share your tracker code?

LilyGO, 2021-10-19:
Hello, have you solved your issuse? I will close this issuse

Rax2647, 2022-02-03:
@loctracker If you could share your code it would be extremely helpful, im trying to get a simple tracker going with adafruit and im pulling my hair out.. i'm connected to the internet fine, gps works fine, however im having nothing but problems converting that code to work.  mainly at the moment i get mqtt and fona not defined.  also an issue with the tinygsm library vs the fona...

mithunspecop, 2023-01-13:
> I worked on some code for a MQTT tracker. It works fine but when I switch from USB power to 18650 battery power, it doesn't seem to work. To be clear, I turn the power on with the switch with only [...]

Hey! have you found any solution? I am facing the similar issue!

DevinCarpenter, 2023-02-09:
Hi guys, have you tried writing logs to a file?

That's how I do all of my debugging when I'm on battery power. Catching error and sending those to a file may be helpful as well so you can see what's happening.

Rax2647, 2023-02-09:
I ended up adapting the code to run sim and wifi and then i telnet into the sim7000 to get the Serial Monitor.. i have the code around here somewhere I'll post it when i find it

Rax2647, 2023-02-09:
So i found the code it's three files that follow...

0TA_Template_Sketch_TelnetStream.ino

''''
> #define ESP32_RTOS  // Uncomment this line if you want to use the code with freertos only on the ESP32
                    // Has to be done before including "OTA.h"

#include "OTA.h"
#include <Credentials.h>

uint32_t entry;

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");

  setupOTA("TemplateSketch", mySSID, myPASSWORD);

  // Your setup code
}

void loop() {
  entry = micros();
#ifdef defined(ESP32_RTOS) && defined(ESP32)
#else // If you do not use FreeRTOS, you have to regulary call the handle method.
  ArduinoOTA.handle();
#endif
  TelnetStream.println(micros()-entry);
  TelnetStream.println("Loop");
  delay(1000);
  // Your code here
Serial.print("Hello World!);
delay(1000);
  
}
''''

Rax2647, 2023-02-09:
OTA.h

'''
#ifdef ESP32
#include <WiFi.h>
#include <ESPmDNS.h>
#else
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#endif

#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <TelnetStream.h>

#if defined(ESP32_RTOS) && defined(ESP32)
void ota_handle( void * parameter ) {
  for (;;) {
    ArduinoOTA.handle();
    delay(3500);
  }
}
#endif

void setupOTA(const char* nameprefix, const char* ssid, const char* password) {
  // Configure the hostname
  uint16_t maxlen = strlen(nameprefix) + 7;
  char *fullhostname = new char[maxlen];
  uint8_t mac[6];
  WiFi.macAddress(mac);
  snprintf(fullhostname, maxlen, "%s-%02x%02x%02x", nameprefix, mac[3], mac[4], mac[5]);
  ArduinoOTA.setHostname(fullhostname);
  delete[] fullhostname;

  // Configure and start the WiFi station
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  // Wait for connection
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  // Port defaults to 3232
  // ArduinoOTA.setPort(3232); // Use 8266 port if you are working in Sloeber IDE, it is fixed there and not adjustable

  // No authentication by default
  // ArduinoOTA.setPassword("admin");

  // Password can be set with it's md5 value as well
  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3");

  ArduinoOTA.onStart([]() {
	//NOTE: make .detach() here for all functions called by Ticker.h library - not to interrupt transfer process in any way.

    String type;
    if (ArduinoOTA.getCommand() == U_FLASH)
      type = "sketch";
    else // U_SPIFFS
      type = "filesystem";

    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
    Serial.println("Start updating " + type);
  });
  
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("\nAuth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("\nBegin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("\nConnect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("\nReceive Failed");
    else if (error == OTA_END_ERROR) Serial.println("\nEnd Failed");
  });

  ArduinoOTA.begin();
  TelnetStream.begin();

  Serial.println("OTA Initialized");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

#if defined(ESP32_RTOS) && defined(ESP32)
  xTaskCreate(
    ota_handle,          /* Task function. */
    "OTA_HANDLE",        /* String with name of task. */
    10000,            /* Stack size in bytes. */
    NULL,             /* Parameter passed as input of the task */
    1,                /* Priority of the task. */
    NULL);            /* Task handle. */
#endif
}
'''

Rax2647, 2023-02-09:
Credentials.h

- [ #pragma once
const char* mySSID = "";
const char* myPASSWORD = "";
]

DevinCarpenter, 2023-02-14:
Nice, glad to hear you solved your problem!