Xinyuan-LilyGO/LilyGO-T-SIM7000G #109: Micro Python Doc error.

TravisDuck999, 2021-04-14:
Hi.

I just was messing around with Micro Python because I could not get any of the C++ libraries to do a POST corretly.
( Last resort ) 

Anyway, I found the instructions on this page : 
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/tree/master/examples/MicroPython_LoBo

To be incorrect, small change, with the "--baud BAUD" ( I did replace BAUD with 9600 ) part of the command.

Just removed that and it connected properly, but with it, the board never connects.

Hope this saves someone some frustration in the future.

Mr-HaleYa, 2021-05-06:
What??? if you fixed a bug submit a PR or describe it better and I will. 

this is an _ISSUES_ board, not a comment section

TravisDuck999, 2021-05-07:
Sorry about that, I'm very new to GitHub and didnt know that was the way things where done, I will do it now.

closed 2021-05-07 by TravisDuck999