Xinyuan-LilyGO/LilyGO-T-SIM7000G #111: Issues with some boards

ejri, 2021-04-16:
Hey,

I just bought 5 boards and have 3 from a previous order. 

Only 2 of them actually connect to an LTE-M cat network, using sim cards from two different companies that specifically set their preferred networks to be LTE-M cat. 

Of the 6 others (all 5 new and 1 old), only 3 connect to a GSM network after like 30s of trying to find a network. (using the Arduino_NetworkTest example)

Resetting the modem doesn't work (using the reset modem example from TinyGSM and shared in this repo).

Any thoughts??? are these boards unreliable?

nautalert, 2021-04-21:
It looks like I'm having the same issue with both V1 and V2 versions of the board. GPS works fine but unable to get a registered connection.

TryingToGITthis, 2021-04-21:
It looks like I'm having the same issue with both V1 and V2 versions of the board. GPS works fine but unable to get a registered connection.

Mr-HaleYa, 2021-05-06:
if you want to work with me I can help you fix this as I have like 30+ boards and all of them work perfectly 
just have to get the code right, let me know

ejri, 2021-05-06:
Sure, thanks. 

I'm mostly using the Azure example I added in this repo. 

The issue is that the modem can not find any networks in some boards. For other boards, the modem does not start at all. 

This is the error message when I try NetworkTest
[image]

I've tried updating the firmware, which didn't work.