Xinyuan-LilyGO/LilyGO-T-SIM7000G #114: Stuck at Info: Initializing SNTP

kedzior-io, 2021-05-10:
I'm using the Azure IOT example as is.

It gets stuck at (even if `WiFi.begin();` is called)

```
Step 10: IMEI: DST: 1
Modem Connected to Rogers' LTE (channel--> 38) CAT-M (preferred network). TLE CAT-M OK
 > Modem On --> True
 > IoT Hub
Info: Initializing SNTP

```

vandebuerierune, 2021-06-07:
any updates? I'm having the same issue.

kedzior-io, 2021-06-07:
@vandebuerierune no luck.

vandebuerierune, 2021-06-08:
I have found a fix, In the code there is a bool HasIotHub which is initially set on false, later on it would be set on true if the iothub connectionstring is correct. However this just doesn't work for some reason. The problem was fixed when I set this boolean on true by default. My code is sending data to iothub right now and it works fluently. Right now I'm doint this in the loop just to make sure that it isn't put to false by accident again.
[image]

I also disabled the shutdown function at the bottom of the loop since shutting down and restarting the module takes to much time. By disabling this it now sends data at a much faster rate.
[image]

I hope this solution helps you and others aswell!

-- Rune Vandebuerie

kedzior-io, 2021-06-08:
@vandebuerierune I haven't had time to go through it and didn't even noticed this variable but hey I fired up with what you suggested and it seems to work. 

Awesome! Thanks!

Though now I'm getting. 

`Failed DNS lookup for my-test.azure-devices.net: 202`

Did you have the same problem?

kedzior-io, 2021-06-09:
Closing this issue since it is solved and opened a new one referencing DNS problems:
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/120

closed 2021-06-09 by kedzior-io