Xinyuan-LilyGO/LilyGO-T-SIM7000G #116: TinyGSM library changed

capedra, 2021-05-22:
Suddenly my code stopped working and I found out some changes on TinyGSM library which works different now.

This part does NOT work anymore:

```
  /*
    2 Automatic
    13 GSM only
    38 LTE only
    51 GSM and LTE only
  * * * */
  String res;
  do {
    res = modem.setNetworkMode(38);
    delay(500);
  } while (res != "OK");

  /*
    1 CAT-M
    2 NB-Iot
    3 CAT-M and NB-IoT
  * * */
  do {
    res = modem.setPreferredMode(1);
    delay(500);
  } while (res != "OK");

```

Now you have to do:
```
  /*
    2 Automatic
    13 GSM only
    38 LTE only
    51 GSM and LTE only
  * * * */
  bool res;

  do {
    res = modem.setNetworkMode(2);
    delay(500);
  } while (!res);

  /*
    1 CAT-M
    2 NB-Iot
    3 CAT-M and NB-IoT
  * * */
  do {
    res = modem.setPreferredMode(3);
    delay(500);
  } while (!res);

```

capedra, 2021-05-23:
```
/*
  2 Automatic
  13 GSM only
  38 LTE only
  51 GSM and LTE only
* * * */
modem.setNetworkMode(2);
modem.waitResponse(GSM_OK);

/*
  1 CAT-M
  2 NB-Iot
  3 CAT-M and NB-IoT
* * */
modem.setPreferredMode(3);
modem.waitResponse(GSM_OK);
```

Mr-HaleYa, 2021-06-15:
congrats, a PR has been made already #119  so this can be closed
they changed setNetworkMode from String to Bool

capedra, 2021-07-21:
@Mr-HaleYa Great! I'm closing this issue.

closed 2021-07-21 by capedra