Xinyuan-LilyGO/LilyGO-T-SIM7000G #117: Not able to flash device

Ollebolle, 2021-05-31:
Hi! I just got this device and it looks awesome, but I'm not able to flash the device at all. I've tried the platformio example as well as simply going through the Arduino IDE but no matter what I do I get this error message:

francoisfanuel, 2021-06-02:
I have the same problem using Big Sur. I switched to platformio remote on a raspi 4 to workaround it until I figure it out.

Mr-HaleYa, 2021-06-15:
yup mac sucks for Arduino development. Use a Windows PC, downgrade the OS, or find another hobby... not much you can do until they fix the bug

Ollebolle, 2021-06-16:
> yup mac sucks for Arduino development. Use a Windows PC, downgrade the OS, or find another hobby... not much you can do until they fix the bug

Yeh I eventually gave up trying to get it to work on Mac and moved over to my Windows machine where it worked right away. 🤷‍♂️

closed 2021-06-16 by Ollebolle