Xinyuan-LilyGO/LilyGO-T-SIM7000G #120: Failed DNS lookup for my-test.azure-devices.net: 202

kedzior-io, 2021-06-09:
Follow up from https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/114

I'm using the Azure IOT example as is.

https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/examples/Arduino_Azure_IoTHub/Arduino_Azure_IoTHub.ino

What I get is:

`Failed DNS lookup for my-test.azure-devices.net: 202`

Mr-HaleYa, 2021-06-15:
No serial log???
No sketch example???
you just expect us to know what this means??? 

> `Failed DNS lookup for my-test.azure-devices.net: 202`

kedzior-io, 2021-06-15:
@Mr-HaleYa 
- serial log - I'm guessing it is useless info here, except the error message, the modem connects correctly and when it tries to reach Azure IOT endpoint (which I have working and accessible from Node-Red already) fails to do DNS lookup. 
- sketch example: it mentions it uses the example as is

kedzior-io, 2021-11-25:
@LilyGO 

It seems it connects OK. though the issue is DNS.  

```
SDCard MOUNT FAIL

Wait...
Initializing modem...
ATE0
AT+CFUN=0

RDY

+CFUN: 1

+CPIN: READY

SMS Ready
AT
AT

OK
ATE0
ATE0

OK
AT+CMEE=0

OK
AT+CLTS=1

OK
AT+CBATCHK=1

OK
AT+CPIN?

+CPIN: READY

OK

DST: 0

*PSUTTZ: 21/11/25,20:34:09","+04",0
Failed to restart modem, attempting to continue without restarting
AT+CFUN=0 

+CPIN: NOT READY

OK
AT+CFUN=1 

OK
Initializing modem...
AT

OK
ATE0

OK
AT+CMEE=0

OK
AT+CLTS=1

OK
AT+CBATCHK=1

OK
AT+CPIN?

ERROR
AT+CPIN?

+CPIN: READY

SMS Ready
AT

+CPIN: READY

OK
ATE0

OK
AT+CMEE=0

OK
AT+CLTS=1

OK
AT+CBATCHK=1

OK
AT+CPIN?

OK

+CPIN: READY

OK
AT+GMM

SIMCOM_SIM7000G

OK
Modem Name: SIMCOM SIM7000G
ATI

SIM7000G R1529

OK
Modem Info: SIM7000G R1529
AT+SGPIO=0,4,1,0

OK
AT+CPIN?

+CPIN: READY

OK
AT+CNMP=38

OK
AT+CMNB=2

OK
AT+CGDCONT=1,"IP","orangeworld","0.0.0.0",0,0,0,0
AT+CGDCONT=13,"IP","orangeworld","0.0.0.0",0,0,0,0

Waiting for network...
AT+CEREG?

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,1

OK
AT+CEREG?

+CEREG: 0,1

OK
Network connected

---Starting GPRS TEST---

Connecting to: orangeworld
AT+CIPSHUT

SHUT OK
AT+CGATT=0

OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=3,1,"APN","orangeworld"

OK
AT+CGDCONT=1,"IP","orangeworld"

OK
AT+CGATT=1

OK
AT+CGACT=1,1

OK
AT+SAPBR=1,1

OK
AT+SAPBR=2,1

+SAPBR: 1,1,"10.169.4.166"

OK
AT+CIPMUX=1

OK
AT+CIPQSEND=1

OK
AT+CIPRXGET=1

OK
AT+CSTT="orangeworld","",""

OK
AT+CIICR

OK
AT+CIFSR;E0

10.169.4.166

OK
GPRS status: AT+CGATT?

+CGATT: 1

OK
AT+CIFSR;E0

10.169.4.166

OK
connected
AT+CCID

8934013372123304107f

OK
CCID: 8934013372123304107f
AT+GSN

869951034364871

OK
IMEI: 869951034364871
AT+COPS?

+COPS: 0,0,"simyo simyo",9

OK
Operator: simyo simyo
AT+CIFSR;E0

DST: 0

*PSUTTZ: 21/11/25,20:34:29","+04",0

10.169.4.166

OK
Local IP: 2785323274
AT+CSQ

+CSQ: 14,99

OK
Signal quality: 14

+CPSI: LTE NB-IOT,Online,214-03,0x4657,218983072,95,EUTRAN-BAND20,6153,0,0,-10,-97,-87,14

OK

---End of GPRS TEST---

AT+CIPSHUT

SHUT OK
AT+CGATT=0

+SAPBR 1: DEACT

OK
AT+CGATT?

+CGATT: 0

OK
GPRS disconnected
AT+CPOWD=1

NORMAL POWER DOWN

````

kedzior-io, 2021-11-25:
Also running this:

https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/examples/Arduino_Azure_IoTHub/Arduino_Azure_IoTHub.ino

goes like that:

```
Started
ESP32 Device
Initializing...
Boot number: 1
Wakeup was not caused by deep sleep: 0
Setup ESP32 to sleep for every 60 Seconds
Waiting till modem ready...
Wait...
AT+IPR=115200

OK
AT

OK
ATE0

OK
AT+CMEE=0

OK
AT+CLTS=1

OK
AT+CBATCHK=1

OK
AT+CPIN?

+CPIN: READY

OK
ATE0

OK
AT+CFUN=0

+CPIN: NOT READY

OK
AT+CFUN=1,1

OK

RDY

+CFUN: 1

+CPIN: READY

SMS ReadyAT

AT

OK
ATE0
ATE0

OK
AT+CMEE=0

OK
AT+CLTS=1

[... 369 lines cut ...]
Error: Time:Thu Jan  1 00:01:25 1970 File:C:\Users\artur\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\AzureIoT\src\az_iot\c-utility\pal\src\tlsio_openssl_compact.c Func:tlsio_openssl_send_async Line:758 tlsio_openssl_send_async without a prior successful open

Error: Time:Thu Jan  1 00:01:25 1970 File:C:\Users\artur\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\AzureIoT\src\az_iot\umqtt\src\mqtt_client.c Func:sendPacketItem Line:332 757: Failure sending control packet data

Error: Time:Thu Jan  1 00:01:25 1970 File:C:\Users\artur\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\AzureIoT\src\az_iot\umqtt\src\mqtt_client.c Func:mqtt_client_disconnect Line:1176 Error: mqtt_client_disconnect send failed

Info: Initializing SNTP

Info: sntp_init

Info: SNTP initialization complete

Info: IoT Hub SDK for C, version 1.1.23

Info: Failed DNS lookup for somenamehere-test.azure-devices.net: 202

Info: >>>Connection status: disconnected

Info: >>>Connection status: disconnected

Info: tlsio_openssl_close has been called when in neither TLSIO_STATE_OPEN nor TLSIO_STATE_ERROR.

Info: Failed DNS lookup for somenamehere-test.azure-devices.net: 202

```

raphac, 2021-12-31:
I have exactly the same problem. First I tried to solve the problem with SNTP and set the time with the help of #https://github.com/vshymanskyy/TinyGSM/issues/357#issuecomment-973873741. After that I had the next error:

```
assertion "Invalid mbox" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c", line 267, function: tcpip_callback_with_block
abort() was called at PC 0x400e4967 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008560c:0x3ffb1c00 0x40085889:0x3ffb1c20 0x400e4967:0x3ffb1c40 0x40101d5a:0x3ffb1c70 0x40101d19:0x3ffb1c90 0x400f81b1:0x3ffb1cd0 0x400d9205:0x3ffb1d20 0x400d5c90:0x3ffb1d80 0x401324e7:0x3ffb1e00 0x400ddc90:0x3ffb1e20 0x400dc50d:0x3ffb1e50 0x400d8fd9:0x3ffb1ec0 0x400d77c1:0x3ffb1ee0 0x400d5299:0x3ffb1f10 0x400d4975:0x3ffb1f60 0x400e2e32:0x3ffb1fb0 0x4008689a:0x3ffb1fd0
  #0  0x4008560c:0x3ffb1c00 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x40085889:0x3ffb1c20 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x400e4967:0x3ffb1c40 in __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)    
  #3  0x40101d5a:0x3ffb1c70 in tcpip_callback_with_block at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:267 (discriminator 1)
  #4  0x40101d19:0x3ffb1c90 in netconn_gethostbyname_addrtype at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/api_lib.c:1057
  #5  0x400f81b1:0x3ffb1cd0 in lwip_getaddrinfo at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/netdb.c:342
  #6  0x400d9205:0x3ffb1d20 in dns_async_is_lookup_complete at C:\Users\rapha\.platformio\packages\framework-arduinoespressif32\libraries\AzureIoT\src\az_iot\c-utility\pal/dns_async.c:118
  #7  0x400d5c90:0x3ffb1d80 in dowork_poll_dns at C:\Users\rapha\.platformio\packages\framework-arduinoespressif32\libraries\AzureIoT\src\az_iot\c-utility\pal\src/tlsio_openssl_compact.c:52
      (inlined by) tlsio_openssl_dowork at C:\Users\rapha\.platformio\packages\framework-arduinoespressif32\libraries\AzureIoT\src\az_iot\c-utility\pal\src/tlsio_openssl_compact.c:695
  #8  0x401324e7:0x3ffb1e00 in xio_dowork at C:\Users\rapha\.platformio\packages\framework-arduinoespressif32\libraries\AzureIoT\src\az_iot\c-utility\src/xio.c:165
  #9  0x400ddc90:0x3ffb1e20 in mqtt_client_dowork at C:\Users\rapha\.platformio\packages\framework-arduinoespressif32\libraries\AzureIoT\src\az_iot\umqtt\src/mqtt_client.c:1205
  #10 0x400dc50d:0x3ffb1e50 in IoTHubTransport_MQTT_Common_DoWork at C:\Users\rapha\.platformio\packages\framework-arduinoespressif32\libraries\AzureIoT\src\az_iot\iothub_client\src/iothubtransport_mqtt_common.c:2512
  #11 0x400d8fd9:0x3ffb1ec0 in IoTHubTransportMqtt_DoWork at C:\Users\rapha\.platformio\packages\framework-arduinoespressif32\libraries\AzureIoT\src\az_iot\iothub_client\src/iothubtransportmqtt.c:115
  #12 0x400d77c1:0x3ffb1ee0 in IoTHubClient_LL_DoWork at C:\Users\rapha\.platformio\packages\framework-arduinoespressif32\libraries\AzureIoT\src\az_iot\iothub_client\src/iothub_client_ll.c:1659
  #13 0x400d5299:0x3ffb1f10 in Esp32MQTTClient_Init at C:\Users\rapha\.platformio\packages\framework-arduinoespressif32\libraries\AzureIoT\src/Esp32MQTTClient.cpp:658
  #14 0x400d4975:0x3ffb1f60 in setup() at C:/Users/rapha/Documents/PlatformIO/Projects/211230-163738-esp32dev/src/Arduino_Azure_IoTHub.ino:524
  #15 0x400e2e32:0x3ffb1fb0 in loopTask(void*) at C:\Users\rapha\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:18
  #16 0x4008689a:0x3ffb1fd0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

When I use Wifi.Begin(), I also get the message: `Failed DNS lookup for xy.azure-devices.net: 202`

Can it be that when trying out the example, a WLAN and password were set in the code and it only worked via the WLAN?

kedzior-io, 2022-02-03:
@raphac , @LilyGO , @lewisxhe  I'm closing this issue. I complete gave up on Azure IOT Hub. I also have found issues using other boards. Switched to my own hosted MQTT broker  (https://mosquitto.org/) and the device works very smoothly.

closed 2022-02-03 by kedzior-io