Xinyuan-LilyGO/LilyGO-T-SIM7000G #124: newbie question on ttgo sim7000g

tradexsrl, 2021-07-02:
Hi just bought this board , i never used esp32 chips , i came from arduino world , i have installed the json package into arduino ide but in the esp32 board i cannot find this board , do i need to select esp32 wrover moudule? in upload speed? in flash frequency 80 mhz or 40mhz ? in flash mode ?
what is core debug level parameter?
i need to connect grove hx711 sensor and grove dh11 and two microswitch , which are the pins completely free that do not change status also at startup from deep sleep?
how to completely turn off bluetooth and wifi since i will never use?
project needs to runs on 18650 battery and need to last at least 15 days sending data each 4 hours via mqtt
thank's in advance

TravisDuck999, 2021-07-02:
Hi.

I have used the basic ESP32 Dev option to program it and it was fine. The standard setting says 240Mhz that I have not changed.

Upload speed I also used as standard at 921600 but I think that is negotiated when the upload starts so while you may wait a little longer it won’t affect the actual operation ( I think ) 

No Idea about core debug levels, I never changed anything.

For the Free pins you need to be really careful as I have run into this problem myself have a look at the Schematics and the pinouts on the sales website, you should also have a look at the ESP32 module as to make sure there is nothing that gets pulled low or high during boot but can be used for other things when the MCU is up and running. ( These can be found online but a good one is here > https://randomnerdtutorials.com/esp32-pinout-reference-gpios/ ) 

I think Wifi and BLE are off unless you turn them on.

As for your battery life, I think you are going to be disappointed, 15 days is a very long time for one battery. I suggest at least having a solar panel on there, you can have a look on here for solar details, lots of info.

I have a project that transmits every 30min and records every 1 min, ( deep sleep in-between ) it lasts 2.5 days.
While I don’t know your application I think you may want to rethink the battery side of things.

My plan for a longer battery life is to buy an 18650 battery pack (3.7V), remove the rear 18650 clip and solder the bigger pack onto the terminals directly.

The battery pack I am using will have 3 x 18650's in parallel so it will take ages to charge fully, but I don’t mind because of the extra battery life.
( The battery pack also will have its own battery managment and protection, while its not needed its just an extra bonus )

tradexsrl, 2021-07-02:
Thank you for the advice , for the battery I imagined that one battery was not enough since I came from arduino hardware and for the same project I’m using 4 18650 battery in parallel and I’m getting 10 days max , that’s why I’m trying with esp chip to see how it goes
Cycle is: power on from deep sleep , reading gps coordinate , reading temperature and humidity from dht11 , reading weight from load cell , reading battery voltage , send these value via MQTT and check if there will be an ota upgrade , then go to deep sleep for 2 or 4 hours
If one of the 2 tamper switch trigger an alert , wake up via interrupt and send an sms

Inviato da iPhone

TravisDuck999, 2021-07-02:
Ah ok, well you might have a chance with only recording every 2 or 4 hours.
I miss-read your initial recording times, I thought it was every 15 min, sorry.

I still suggest trying to put a small solar panel on there if at all possible.

Good luck, close this thread if you are happy with this info though.

Cheers