Xinyuan-LilyGO/LilyGO-T-SIM7000G #127: New Board Arduino Code uploading

ikequan, 2021-07-14:
Last month I purchased 5 boards from the Aliexpress link and being trying to upload sketch but with no success. I did some search found out others are facing the same issue with new products from Xinyuan-LilyGO. The boards I purchased last are still working well. Can someone help figure this issue. I've tried updating all drivers but the problem persists. below are some logs. Thank you.

Arduino: 1.8.13 (Mac OS X), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

Sketch uses 220166 bytes (16%) of program storage space. Maximum is 1310720 bytes.
Global variables use 13440 bytes (4%) of dynamic memory, leaving 314240 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.0-dev
Serial port /dev/cu.usbmodem52780019601
Connecting....
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:6f:28:d0:1e:34
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000)
A fatal error occurred: Failed to write to target RAM (result was 01070000)

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

rnborland, 2021-07-30:
ikequan, I have had a couple of issues with this, #1  Make sure the sd card is removed before uploading.  #2  I have had issues with certain GPIO pins interfereing with code uploads, disconnect GPIO 4 and possibly 2 prior to uploading, you can reconnect them after the upload is completed.
Regards

LilyGO, 2021-09-25:
Are you able to upload normally now? If not, please feel free to contact

LilyGO, 2021-10-19:
It looks like a similar [problem](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/139) , and it's been solved

closed 2021-10-19 by LilyGO