Xinyuan-LilyGO/LilyGO-T-SIM7000G #128: read battery voltage when usb connected for code debugging

tradexsrl, 2021-07-19:
Hi , i have read the battery voltage from IO35 is not possible when the usb is connected to a pc but what about when you are debugging your code into arduino ide or platformio and needs to do some serial.print?
is also impratical to implement some mqtt to send that value , then load the sketch , then disconnect and see what happend...and repeat...if somethings goes wrong into the gprs/mqtt part you are stuck with more doubt than before

procopio, 2021-10-08:
Never tried but you can read the battery from SIM7000 using AT commands.

tradexsrl, 2021-10-19:
@LilyGO , sorry but i don't understand....can you elaborate? where i need to connect gnd - tx - rx to the board ? and then what i need to write in the sketch? at what speed ? etc etc

LilyGO, 2021-10-19:
Perhaps a Google search would help you understand the tool?

tradexsrl, 2021-10-19:
thank you for the clever reply.... ;-) i can do it but how you think i can find the specific explanation for this board and step by step instruction ? this is why i'm asking directly to the Lilygo Factory....