Xinyuan-LilyGO/LilyGO-T-SIM7000G #130: Cannot reset SIM7000G module

pixelfriese, 2021-07-31:
Hi. 

when I run the reset program, I get the following output. The baud rate is set correctly. I can't send AT commands to the board from the console. Only when I insert the AT command in the program code will they run. My guess is that I won't get any feedback on the if (SerialAT.available) query if the modem is already running.:

```
14:25:32.941 -> 
14:25:32.941 -> Starting Up Modem...
14:25:32.941 -> 
14:25:32.975 -> Testing Modem Response...
14:25:32.975 -> 
14:25:32.975 -> ****
14:25:34.437 -> u⸮
14:25:36.443 -> AT

14:25:36.477 -> OK
14:25:36.477 -> 
14:25:36.477 -> ****
14:25:36.477 -> 
14:25:36.477 -> AT&FZE0&W
14:25:36.545 -> AT&FZE0&W

14:25:36.613 -> Autobaud support: (9600,19200,38400,57600,115200).
14:25:36.613 -> 
14:25:36.613 -> OK
14:25:36.613 -> AT+IPR=0
14:25:37.567 -> AT+IFC=0,0
14:25:38.587 -> AT+ICF=3,3
14:25:39.573 -> AT+CSCLK=0
14:25:40.593 -> AT&W
14:25:41.751 -> 
14:25:41.751 -> ***********************************************************
14:25:41.751 ->  Reset settings to Factory Default: FAIL
14:25:41.751 -> ***********************************************************

```

But if I press the reset button on the board again and the modem is already active, I get the following output:

```
14:32:37.269 -> Starting Up Modem...
14:32:47.517 -> 
14:32:47.555 -> Testing Modem Response...
14:32:47.555 -> 
14:32:47.555 -> ****
14:32:57.745 -> ****
14:32:57.745 -> 
14:32:57.745 -> ***********************************************************
14:32:57.745 ->  Failed to connect to the modem! Check the baud and try again.
14:32:57.745 -> ***********************************************************
14:32:57.745 -> 
```

_Originally posted by @pixelfriese in https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/38#issuecomment-890342296_

ceren16, 2023-08-02:
> Yes, I was able to solve it. The AT commands from the reset script had to be executed manually. After that the module could be used normally again.

hello can you send code ? thanks you

ceren16, 2023-08-03:
> Hi,
> 
> Yes, I was able to solve it. The AT commands from the reset script had to be executed manually. After that the module could be used normally again.

hello can you send code