Xinyuan-LilyGO/LilyGO-T-SIM7000G #132: OpenOCD cannot access the board - jtag_trst == 0 assertion failed

eisenach, 2021-08-11:
Hello,
I'm trying to debug my firmware on TTGO-T-SIM7000G through OpenOCD and, regardless of the driver I'm using, and consistently on Windows and Linux, I have the following result
```
Reading symbols from D:\projects\sss\sss-lock-firmware\.pio\build\esp-wrover-kit\firmware.elf...
done.
PlatformIO Unified Debugger -> http://bit.ly/pio-debug
PlatformIO: debug_tool = ftdi
PlatformIO: Initializing remote target...
Open On-Chip Debugger  v0.10.0-esp32-20210721 (2021-07-21-13:35)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz
WARNING: boards/esp32-wrover.cfg is deprecated, and may be removed in a future release.
         If your board is ESP32-WROVER-KIT, use board/esp32-wrover-kit-1.8v.cfg instead.
adapter speed: 5000 kHz

Info : tcl server disabled
Info : telnet server disabled
Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'
Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6014, description '*', serial '*' at bus location '*'

Assertion failed!

Program: D:\tools\pio\.platformio\packages\tool-openocd-esp32\bin\openocd.exe
File: ../src/jtag/core.c, Line 343

Expression: jtag_trst == 0
.pioinit:11: Error in sourced command file:
Remote communication error.  Target disconnected.: Success.
```

I also tried to run OpenOCD directly (not through PlatformIO), without luck.
Could you please advise?

Thank you,
Dom

closed 2021-10-19 by LilyGO

tolgraven, 2021-12-28:
@LilyGO having the same issue. The default FTDI config clearly isn't the right one, but which one is? Which interface config do we use?

wwestrup, 2023-06-23:
Try adding the following to your platformio.ini file for your project. Make sure to modify the port and debug tool names as needed:

upload_port = /dev/ttyACM0
debug_tool = esp-prog
debug_init_break = tbreak setup
debug_port = /dev/ttyUSB1
build_type = debug
debug_speed = 5000

This will at least get you as far as I've gotten. (but it still doesn't work, at least not for me, yet.)

MateusLourenco97, 2023-07-16:
I got it working using the configurations @wwestrup  mentioned. What could probably going wrong with your peoject is that you're using the LED pin (GPIO 12) on your application code. To get it working you have to give up using the SD slot and the built in LED pins.

shanedoolane, 2023-10-02:
For anyone coming to see this. @MateusLourenco97 appears to have found the solution. If you remove LED control from your code on GPIO12 (the blue LED on our board) and start debugger, it works. 

Otherwise, if you attempt to use the LED and debugger at the same time, it will fail to initialize. 

For ESP-prog, use common GND, 3v3, GPIO12-15. 

```
[env:esp32dev]
platform = espressif32
board = esp-wrover-kit
framework = arduino
debug_tool = esp-prog
debug_init_break = tbreak setup
```

shanedoolane, 2023-10-02:
@LilyGO or @lewisxhe do you know the reason GPIO12 was chosen for LED and JTAG interface? do you know how to make both work together?

shanedoolane, 2023-10-09:
I'm also seeing that I can't use JTAG/prog interface during deep sleep. The USB is able to wake out of deep sleep and successfully boot but JTAG fails.

shanedoolane, 2023-10-09:
Found a solution to wake from deep sleep when getting this error. 
```cpp
#include "driver/rtc_io.h"

rtc_gpio_pullup_en(GPIO_NUM_12);
rtc_gpio_pulldown_dis(GPIO_NUM_12);
esp_sleep_enable_ext0_wakeup(GPIO_NUM_12, 0);
```

This causes GPIO to be pulled high in the application setup. The last line sets a wake-up from deep sleep interrupt on GPIO 12. Whenever GPIO12 is pulled low and in deep sleep, the ESP will wake up. Probing, we see that GPIO 12 is pulled high while app is running and in deep sleep. When we pull it down (i.e. when using JTAG, the ESP is woken from deep sleep).

Here's the logic analyzer trace of the JTAG initial handshake. You can see GPIO12 is pulled high before JTAG starts and as soon as it goes low, the ESP comes out of deep sleep.

[image]

I also tested with the reverse code (i.e. normally low with a pull-up wakeup trigger) but it did not work. Because esp-prog hold GPIO 12 high, the interupt is triggered as soon as the ESP sleeps so it continuously boots. Don't do it this way.
```cpp
#include "driver/rtc_io.h"

rtc_gpio_pullup_dis(GPIO_NUM_12);
rtc_gpio_pulldown_en(GPIO_NUM_12);
esp_sleep_enable_ext0_wakeup(GPIO_NUM_12, 1);
```

[image]