Xinyuan-LilyGO/LilyGO-T-SIM7000G #140: GSM unknown signal connection +CSQ: 99,99

GeorgeSikora, 2021-10-02:
Hello,
Iam trying to send SMS via our module with AT commands, but if I try to get how much signal I have, it just send me unknown signal like "99,99" wich is written in documentation.  After trying to send message, it just send me "+CMS ERROR: 500" ... The pin is unlocked, but nothing wants to work... Can you help me, iam lost :(

[image]

[image]

GeorgeSikora, 2021-10-06:
Here it is, at the beginning it just spams "AT+CEREG?", after some while it shows few "+CGREG: 0,2" but normally it response "+CGREG: 0,0", so I dont know where is the problem ?

_The console:_
```

AT+CEREG?
+CEREG: 0,0
OK
AT+CGREG?
+CGREG: 0,0
OK
AT+CEREG?
+CEREG: 0,0
OK
AT+CGREG?
+CGREG: 0,0
OK
Initializing modem...
AT
OK
ATE0
OK
AT+CMEE=0
OK
AT+CLTS=1
OK
AT+CBATCHK=1
OK
AT+CPIN?
+CPIN: READY
OK
AT+GMM
SIMCOM_SIM7000G
OK
Modem Name: SIMCOM SIM7000G
ATI
SIM7000G R1529
OK
Modem Info: SIM7000G R1529
AT+SGPIO=0,4,1,0
OK
AT+CPIN?
+CPIN: READY
OK
AT+CPIN="3245"
ERROR
AT+CNMP=38
OK
AT+CMNB=2
OK
AT+CGDCONT=1,"IP","internet","0.0.0.0",0,0,0,0
AT+CGDCONT=13,"IP","internet","0.0.0.0",0,0,0,0
Waiting for network...
AT+CEREG?
OK
AT+CGREG?
+CGREG: 0,0
OK
AT+CEREG?
+CEREG: 0,0
OK
AT+CGREG?
+CGREG: 0,0
OK
AT+CEREG?
+CEREG: 0,0

```

LilyGO, 2021-10-15:
Hello, is your SIM card normal?

GeorgeSikora, 2021-10-15:
Yes, on TTGO esp33 SIM800L it works properly

pploume, 2021-11-23:
Hi,
Same problem ...
stuck at **+CGREG: 0,2** with the above mentionned example.

I've tryied several params for  **modem.setNetworkMode** and **modem.setPreferredMode** ...

SIM from thingsmobile.com, working in my own mobile phone

Signal strength seems OK : **+CSQ : 99,99**
Status Led is always ON
Networks Led is blinking

I'm lost, thanks for your help

platformio.ini as follows : 
[env:esp-wrover-kit]
platform = espressif32
board = esp-wrover-kit
framework = arduino
monitor_speed = 115200
lib_deps = 
	vshymanskyy/TinyGSM@^0.11.4
	vshymanskyy/StreamDebugger@^1.0.1

first lines in terminal : 

Initializing modem...
ATE0
AT+CFUN=0
**Failed to restart modem, attempting to continue without restarting**
AT+CFUN=0
AT+CFUN=1 
AT+CFUN=1
OK
Initializing modem...
AT
AT
OK
ATE0
ATE0
OK
AT+CMEE=0
OK
AT+CLTS=1
OK
AT+CBATCHK=1
OK
AT+CPIN?
+CPIN: READY
OK
AT+GMM
SIMCOM_SIM7000G
OK
Modem Name: SIMCOM SIM7000G
ATI
SIM7000G R1529
OK
Modem Info: SIM7000G R1529
AT+SGPIO=0,4,1,0
OK
AT+CNMP=2
OK
AT+CMNB=3
OK

GeorgeSikora, 2022-01-21:
I has something wrong with board... new works fine.

closed 2022-01-21 by GeorgeSikora

Rasmus105, 2023-11-18:
I am encountering similar problems.

What was wrong with your board?