Xinyuan-LilyGO/LilyGO-T-SIM7000G #145: At&t Cat M Network

Rax2647, 2021-10-29:
Hello,  I just got in the mail 2 lilygo T-SIM7000G boards that I'm trying to connect to the AT&T cat M network..  I'm using the Arduino_NetworkTest sketch and Neither will connect with changing the APN to m2m64.com.attz and changing nothing else.   The boards appear to be working as far as uploading the sketches, I've tried uploading with the arduino ide from my computer and arduino Droid with my phone. The board type I've tried esp32 dev module and wrover module...

In the  Serial monitor i get

18:01:42.205 -> Signal: 30 isNetworkConnected: NO CONNECT
18:01:43.321 -> Signal: 30 isNetworkConnected: NO CONNECT
18:01:44.439 -> Signal: 30 isNetworkConnected: NO CONNECT
18:01:45.555 -> Signal: 30 isNetworkConnected: NO CONNECT
18:01:46.716 -> Signal: 30 isNetworkConnected: NO CONNECT
18:01:47.794 -> Signal: 30 isNetworkConnected: NO CONNECT
18:01:48.869 -> Try 52 method
18:01:51.936 -> Signal: 31 isNetworkConnected: NO CONNECT
18:01:53.055 -> Signal: 31 isNetworkConnected: NO CONNECT
18:01:54.173 -> Signal: 31 isNetworkConnected: NO CONNECT
18:01:55.290 -> Signal: 31 isNetworkConnected: NO CONNECT
18:01:56.409 -> Signal: 31 isNetworkConnected: NO CONNECT
18:01:57.525 -> Signal: 31 isNetworkConnected: NO CONNECT
18:01:58.642 -> Signal: 31 isNetworkConnected: NO CONNECT

The signal bounces between 9 and 32, 99 when it first starts. And I've tried in a few locations so far.. 
[image]
Any help would be appreciated.

Rax2647, 2021-10-29:
Also when i run the AllFunctions.ino changing only the apn i get

[monitor.txt](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/files/7445306/monitor.txt)

Rax2647, 2021-10-29:
Also when i run the AllFunctions.ino changing only the apn i get

`
18:30:15.942 -> AT+CFUN=0
18:30:18.698 -> 怘⸮f`⸮f⸮⸮Failed to restart modem, attempting to continue without restarting
18:30:25.954 -> AT+CFUN=0 
18:30:35.946 -> AT+CFUN=1 
18:30:45.947 -> Initializing modem...
18:30:45.953 -> AT
18:30:46.246 -> AT
18:30:46.257 -> AT
18:30:46.261 -> OK
18:30:46.264 -> ATE0
18:30:46.268 -> ATE0
18:30:46.271 -> OK
18:30:46.273 -> AT+CMEE=0
18:30:46.276 -> 
18:30:46.278 -> OK
18:30:46.280 -> AT+CLTS=1
18:30:46.282 -> 
18:30:46.285 -> OK
18:30:46.288 -> AT+CBATCHK=1
18:30:46.290 -> 
18:30:46.292 -> OK
18:30:46.294 -> AT+CPIN?
18:30:46.297 -> 
18:30:46.299 -> +CPIN: READY
18:30:46.302 -> 
18:30:46.303 -> OK
18:30:46.305 -> AT+GMM
18:30:46.308 -> 
18:30:46.310 -> SIMCOM_SIM7000G
18:30:46.313 -> 
18:30:46.314 -> OK
18:30:46.778 -> Modem Name: SIMCOM SIM7000G
18:30:46.785 -> ATI
18:30:46.789 -> 
18:30:46.791 -> SIM7000G R1529
18:30:46.793 -> 
18:30:46.796 -> OK
18:30:47.288 -> Modem Info: SIM7000G R1529
18:30:47.296 -> AT+SGPIO=0,4,1,0
18:30:47.300 -> 
18:30:47.303 -> OK
18:30:47.305 -> AT+CPIN?
18:30:47.308 -> 
18:30:47.310 -> +CPIN: READY
18:30:47.314 -> 
18:30:47.316 -> OK
18:30:47.318 -> AT+CNMP=38
18:30:47.321 -> 
18:30:47.323 -> OK
18:30:47.325 -> AT+CMNB=2
18:30:47.328 -> 
18:30:47.330 -> OK
18:30:48.808 -> AT+CGDCONT=1,"IP","m2m64.com.attz","0.0.0.0",0,0,0,0
18:30:48.818 -> AT+CGDCONT=13,"IP","m2m64.com.attz","0.0.0.0",0,0,0,0
18:30:48.827 -> 
18:30:48.830 -> 
18:30:48.833 -> 
18:30:48.835 -> Waiting for network...
18:30:48.838 -> AT+CEREG?
18:30:48.842 -> 
18:30:48.844 -> OK
18:30:48.853 -> 
18:30:48.861 -> OK
18:30:48.868 -> 
18:30:48.871 -> +CEREG: 0,0
18:30:48.874 -> 
18:30:48.876 -> OK
[... 240 lines cut ...]
18:30:55.154 -> AT+CGREG?
18:30:55.161 -> 
18:30:55.164 -> +CGREG: 0,2
18:30:55.168 -> 
18:30:55.172 -> OK
18:30:55.382 -> AT+CEREG?
18:30:55.397 -> 
18:30:55.409 -> +CEREG: 0,0
18:30:55.413 -> 
18:30:55.418 -> OK
18:30:55.422 -> AT+CGREG?
18:30:55.426 -> 
18:30:55.430 -> +CGREG: 0,2
18:30:55.441 -> 
18:30:55.453 -> OK
18:30:55.643 -> AT+CEREG?
18:30:55.661 -> 
18:30:55.667 -> +CEREG: 0,0
18:30:55.671 -> 
18:30:55.675 -> OK
18:30:55.679 -> AT+CGREG?
18:30:55.683 -> 
18:30:55.689 -> +CGREG: 0,2
18:30:55.696 
`

Rax2647, 2021-11-01:
When i use the Reset Modem code example under [https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/examples/Arduino_TinyGSM/ResetModem/ResetModem.ino](url)  i get a reurn of 

`19:51:15.942 -> ⸮⸮⸮⸮⸮<⸮⸮&ki*jp⸮⸮⸮q⸮aNqja
19:51:17.408 -> Starting Up Modem...
19:51:27.718 -> 
19:51:27.718 -> Testing Modem Response...
19:51:27.751 -> 
19:51:27.751 -> ****
19:51:37.721 -> ****
19:51:37.721 -> 
19:51:37.721 -> ***********************************************************
19:51:37.787 ->  Failed to connect to the modem! Check the baud and try again.
19:51:37.853 -> ***********************************************************
19:51:37.921 -> 
19:52:41.362 -> ⸮⸮⸮⸮⸮<⸮⸮&ki*jp⸮⸮⸮q⸮aNqja
19:52:42.862 -> Starting Up Modem...
19:52:53.163 -> 
19:52:53.163 -> Testing Modem Response...
19:52:53.198 -> 
19:52:53.198 -> ****
19:53:03.166 -> ****
19:53:03.166 -> 
19:53:03.166 -> ***********************************************************
19:53:03.232 ->  Failed to connect to the modem! Check the baud and try again.
19:53:03.299 -> ***********************************************************
19:53:03.365 -> 
19:59:40.239 -> ⸮⸮⸮⸮⸮<⸮⸮&kI
19:59:40.272 -> jP⸮⸮⸮QҚanqJa
19:59:41.738 -> Starting Up Modem...
19:59:52.036 -> 
19:59:52.036 -> Testing Modem Response...
19:59:52.070 -> 
19:59:52.070 -> ****
20:00:02.034 -> ****
20:00:02.034 -> 
20:00:02.034 -> ***********************************************************
20:00:02.101 ->  Failed to connect to the modem! Check the baud and try again.
20:00:02.167 -> ***********************************************************
20:00:02.233 -> 
`

Rax2647, 2021-11-01:
And just for further information i tried switching the antenna out to the pulse antenna and no change, also this simcard works and publishes to adafruit.io in the 

[DIYmalls SIM7000A Board 4G GSM... https://www.amazon.com/dp/B0825MFP8P?ref=ppx_pop_mob_ap_share](url)

LilyGO, 2021-11-01:
Hello. If you are in the US, you should buy sim7000A instead of sim7000G

Rax2647, 2021-11-01:
Shouldn't the G work as well as the A, the G supports the same bands no?

[https://github.com/botletics/SIM7000-LTE-Shield/blob/master/SIM7000%20Documentation/Spec%20Overview%20Docs/SIM7000G_SPEC_2018-05-15.pdf](url)

[https://github.com/botletics/SIM7000-LTE-Shield/blob/master/SIM7000%20Documentation/Spec%20Overview%20Docs/SIM7000A_SPEC_2017-9-21.pdf](url)

Rax2647, 2021-11-01:
> Hello. If you are in the US, you should buy sim7000A instead of sim7000G

https://github.com/botletics/SIM7000-LTE-Shield/wiki/Board-Versions

LilyGO, 2021-11-01:
Only SIM7000A has AT&T and Version authentication.

bhupiister, 2021-12-26:
@LilyGO Can sim7000G work in USA with any service provider such as hologram or Google sim ? I am asking because sim7000G is easily available. 
@Rax2647 were you able to solve?

Rax2647, 2021-12-26:
> @LilyGO Can sim7000G work in USA with any service provider such as hologram or Google sim ? I am asking because sim7000G is easily available. 
> @Rax2647 were you able to solve?

I was by altering the network bands on the network test sketch, i was able to make it read connected and give location what not through at commands.  I'll upload the code later, I'm on my way to work. My current issue is getting the gps to relay to Adafruit.io as my project is a gps tracker. I'm not the best at coding and I'm having issues with the Adafruit fona library stepping over the tiny gsm library i think...

Rax2647, 2021-12-26:
>     Serial.println("=====Preferred mode selection=====");
    modem.sendAT("+CNMP?");
    if (modem.waitResponse(1000L, res) == 1) {
        res.replace(GSM_NL "OK" GSM_NL, "");
        Serial.println(res);
    }
    res = "";
    Serial.println("=======================");

    Serial.println("=====Preferred selection between CAT-M and NB-IoT=====");
    Serial.println("ADDED BAND ALTERATION...");
     modem.sendAT("+CBANDCFG=2,4,12,13 ");
    modem.sendAT("+CMNB=1");
    if (modem.waitResponse(1000L, res) == 1) {
        res.replace(GSM_NL "OK" GSM_NL, "");
        Serial.println(res);
    }
    res = "";
    Serial.println("=======================");
<

Rax2647, 2021-12-26:
> @LilyGO Can sim7000G work in USA with any service provider such as hologram or Google sim ? I am asking because sim7000G is easily available. 
> @Rax2647 were you able to solve?

Now if you have an example sketch to adafruit io or a similar mqtt site that's preferably free and allows for map tracking and subscribe for three switches.  . I tried adapting the Botletics sketch to this one and partial luck so far.. they are very conflicting due to the library's i believe, i could be wrong though if i actually took a coding class

bhupiister, 2021-12-27:
I'll rather be pushing data on my own server. All I want is to be able to send small string which is acceleration data to my server every 5 minutes. My project does not use GPS. So if you were able to connect to internet, I can finally buy and test it.

bhupiister, 2021-12-27:
> Now if you have an example sketch to adafruit io or a similar mqtt site that's preferably free and allows for map tracking and subscribe for three switches. . I tried adapting the Botletics sketch [...]

I generally don't use any library and make my own functions to send and receive AT commands. It is very easy to follow and not hard to code. Just make sure that you read the updated document from SIMCOM and follow the commands. I can help you once I have this module in my hand.

Rax2647, 2022-01-25:
@bhupiister Thank you, it would be great if you could just get me on my feet transferring a working code i had for another board for this one, again my coding knowledge is very limited...   I created a repository where i added my working code for the lilygo t sim7000g as well as my old code for the sim7000 evb board and esp8266, and a butchered mix of the two, any help would be appreciated

Rax2647, 2022-01-25:
@bhupiister   https://github.com/Rax2647/GPS-Tracker.git

Rax2647, 2022-02-22:
Solved !