Xinyuan-LilyGO/LilyGO-T-SIM7000G #146: can't power down modem

andysc, 2021-11-08:
I'm trying to power down the SIM7000 before putting the ESP32 into deep sleep.

I tried pulling pin 4 low, which sometimes works, but not always.
I tried AT+CPOWD (per the manual) but that isn't allowed.
I tried AT+CPOWD=1 (UTSL from TinyGSM), and that shuts down the modem (yay!), but 5 seconds later it restarts.

I have tried keeping the ESP awake, so nothing changes on the pins - modem still restarts after 5 seconds.
I tried doing AT+CPOWD=1 and then pull pin 4 low, but it still restarts after 5 seconds.

I have shutdown the GPS and GPRS before I attempt to power down (in case that was keeping it alive).

How can I properly shut down a SIM7000G ?

andysc, 2021-11-08:
Looks like https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/144 is having a similar issue.
I tried the 300ms high, followed by low. No difference.

Seems we're getting into chicken-waving territory here, though. 
The SIM7000 spec says at least 1200ms low, followed by high, but with no upper limit on the low. 
So that would be compatible with the MCU going to sleep and the pins staying low.

LilyGO, 2021-11-09:
[reference](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/examples/Arduino_TinyGSM/AllFunctions/AllFunctions.ino)
 
Line 310     modem.sendAT("+CPOWD=1");   //Shut down the 7000G module

//Wake up regularly
Line 318     esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
                  delay(200);
                  esp_deep_sleep_start();

andysc, 2021-11-09:
Thanks - so you send AT+CPOWD=1 twice, then - once at line 310, and then that is also what modem.poweroff() does in line 314.

I am not running the "wake up regularly" part of the code (it's my own application, not this one).

I'll try the CPOWD twice trick to see if that helps :)
Thank you

closed 2021-11-11 by andysc