Xinyuan-LilyGO/LilyGO-T-SIM7000G #147: Auto-restart device when using 18650 and solar panel.

Eckaard, 2021-11-22:
Hi all,

So I've written a piece of code, in micropython, that measures distance using an ultra sonic sensor and then sends the reading to a influxdb instance. My issue is as follows:  Once the battery has drained to completion, the device needs to be hard reset in order to work once the battery has been charged up again. The device is powered using an 18650 and charged using a 5V 750mA solar panel.

I've tried using a "try" "except", however, the device seems to be hanging in deepsleep.

This does, however, not occur when charging the device using an usb-c cable. The device resets itself and starts posting data straight away without a hard reset.

Any advice would be highly appreciated.

RafBois, 2022-03-16:
Hello, did you find a solution?

Best regards

Eckaard, 2022-03-16:
Evening,

Unfortunately I did not. I haven't worked on this project in quite some time. What you could maybe look at doing is using a powerbank. This way the solar panel will charge the powerbank which will only supply power, via USB C, to the t-sim 7000 once it's been charged to a sufficient amount. This, however, isn't really a proper solution. Please let me know if you do figure something out

> Hello, did you find a solution?
> 
> Best regards

RafBois, 2022-03-17:
I don't have the material yet but I will try to reproduce this problem which seems to me major for my project.
I will try to find a solution.

Thanks for your feedback

LilyGO, 2022-03-21:
Hello, does this happen with the example?

nepat, 2022-06-28:
I have experienced similar power instability issues. In particular in GSM mode the SIM module temporarily draws a large current. This then causes voltage to drop on the battery pins by a fairly substantial level (around 0.6V). Depending on the state of charge, the ESP32 can then fall into an unstable operating mode. My solution has been to avoid GSM whenever possible and operate the device in LTE-only mode. Also I charge the battery with an external solar charger module only, as that a) allows optimized charging at the proper power point (pseudo - MPPT) for a 6V module and b) can handle the power from a 6W module without overheating.

Mr-HaleYa, 2022-07-29:
buy a hardware watchdog and use that. Or desolder the battery socket and attach an external battery pack with an inline BMS that has a slightly higher cut-off voltage

I've done both

My best way of fixing this was to design an expansion board that these sockets into and the expansion board is essentially a hardware watchdog.

---
a snippet of my fix 

[image]
[image]

fgnievinski, 2022-12-17:
> I have experienced similar power instability issues. In particular in GSM mode the SIM module temporarily draws a large current. This then causes voltage to drop on the battery pins by a fairly [...]

a capacitor in parallel will take care of the spikes in current drawn.