Xinyuan-LilyGO/LilyGO-T-SIM7000G #148: Antenna/Band trouble ?

pploume, 2021-11-24:
Hi,

Trying with example "all-functions.ino", no success.
The program loops with "+CGREG: 0,2"

The Signal Strength is 99,99
I have this message : Failed to restart modem, attempting to continue without restarting

Tested with 2 SIM cards, 1 from SFR, 1 from ThingsMobile.com
(both OK when in my mobile phone)

There is only one antenna around, it shows :
UMTS 880.1 à 889.9 MHz
UMTS 925.1 à 934.9 MHz
LTE 791 à 801 MHz
LTE 832 à 842 MHz

the problem is here ?

I can provide more info on request

Thanks for your help

LilyGO, 2021-11-25:
Hello. Check your SIM card and check with your local carrier to see if they support cat-M/NB-iot /GSM networks .
 Refer to article  [9](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G#notes-for-new-users).

pploume, 2021-11-25:
Hi,
Thanks for your answer ...

1°) "refer to article 9" : done, all seems to be OK
2°) "check your SIM card" :
      - SIM card works in my own phone 
      - sim provider "thingsmobile" support CAT-M (and actually not NB-iot)
      
3°) you mention "GSM networks" ...
- do you mean 2G ?
- It seems thant there is only UMTS + LTE, as mentionned previously

4°) Modem Info: SIM7000G R1529 
- need modem upgrade ?

thanks for your help

LilyGO, 2021-12-01:
3.SIM7000G supports CAT-M, NB-iot, and GSM.If your SIM card only supports CAT-M.Refer to article 9 for correct configuration. （From 1 and 2, we know that you are running normally）
4. You can send AT+SIMCOMATI to view the current version information