Xinyuan-LilyGO/LilyGO-T-SIM7000G #151: New firmware 1529B08SIM7000G

dzungpv, 2021-12-26:
Hello, I see new firmware here : https://techship.com/downloads/category/firmware-simcom/ but i could not download.
@lewisxhe cann you download it and share us?

LilyGO, 2021-12-28:
https://drive.google.com/drive/folders/16V0M37zw5FSplnWWDiyBxrzjP-fbgAd6

cuplesoft, 2021-12-28:
I tried flash my module and install new firmware by soldered usb. First time with success. But i decided to flash again and during installation somethings went wrong. 
Installation stops and after that i disconnected usb cable. It is mean my SIM7000G is dead from now definitely? Not responding...

emielch, 2022-03-31:
Thanks for sharing the new firmware!
Could you also share the release notes so we know what got changed?

LilyGO, 2022-04-06:
[1529SIM7000G Firmware Release Note.pdf](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/files/8422838/1529SIM7000G.Firmware.Release.Note.pdf)