Xinyuan-LilyGO/LilyGO-T-SIM7000G #154: modem.setGNSSMode ERROR

FabPari, 2022-01-17:
Hi Dear
I try to test your example file but all give me an error, in particular  Arduino_TinyGSM/AllFunctions/ return me an error when I try to SET GNSS Mode
modem.setGNSSMode(2, 1);
I try all the possible combinations, but I hallways receive an output ERROR to serial output.
Can you help me to understand why?

`6:53:15.270 -> AT+CNMP=2
16:53:15.270 -> 
16:53:15.270 -> OK
16:53:15.270 -> [984] setNetworkMode: 1
16:53:15.270 -> AT+CGNSSMODE?
16:53:15.304 -> 
16:53:15.304 -> +CGNSSMODE: 1
16:53:15.304 -> 
16:53:15.304 -> OK
16:53:16.291 -> [2026] GNSS Mode: 1
16:53:17.316 -> AT+CGNSSMODE=2,1
16:53:17.316 -> 
16:53:17.316 -> ERROR
16:53:17.316 -> AT+CGMM
16:53:17.316 -> 
16:53:17.316 -> A7670E-LASE
`
After this output start a loop

`16:53:17.350 -> +CGREG: 0,0
16:53:17.350 -> 
16:53:17.350 -> OK
16:53:17.588 -> AT+CGREG?
`

No output info about lat and long..... nothing
Can you help me?
Thanks a lot
Fabio

LilyGO, 2022-01-18:
Hello, what module are you using? SIM7000G?

FabPari, 2022-01-18:
Hi Dear I really don't understand, on PCB there is written SIM7000X but on the chip there is A7670E
Can you help me?
Thanks
Fabio

LilyGO, 2022-01-19:
Can you send me the product picture?

FabPari, 2022-01-24:
[image]

LilyGO, 2022-01-25:
A7670-LASE No GNSS functionality.  There will soon be a new version of the A7670 board, GNSS functionality will be optional

FabPari, 2022-01-25:
Why there is a GPS connector and it's sell with GPS antenna?

LilyGO, 2022-01-25:
Reference https://github.com/Xinyuan-LilyGO/LilyGO-T-A7670X
GNSS (GPS) is optional.  The A7670 package is compatible with SIM7000.  It looks the same as the SIM7000 board.  
[image]