Xinyuan-LilyGO/LilyGO-T-SIM7000G #158: Need help with servo

MichaelLeipzig, 2022-02-08:
Hello,

I know, it sounds easy, but I can´t get it to work. I want to use a servo (e. g. SG90). All other components (relay, sensors, MQTT, GPS) work fine. But I broke 3 servos now. All of them struggeled one time at the "setup" and were broken afterwards.

I wired like this:

T-SIM7000G:
VIN -> external 5V-Power-Supply
GND -> external 5V-Power-Supply
GPIO27 -> Servo

Servo SG90:
VIN -> same external 5V-Power-Supply
GND -> same external 5V-Power-Supply
yellow -> GPIO27

At first, I tried my own coding, then I used the example-sketch for the ESP32-Servo-Library (this workes with an ESP32-Wrover-Module and the same servo):

`#include <ESP32Servo.h>
int APin = 16;
ESP32PWM pwm;
int freq = 1000;
void setup() {
	// Allow allocation of all timers
	ESP32PWM::allocateTimer(0);
	ESP32PWM::allocateTimer(1);
	ESP32PWM::allocateTimer(2);
	ESP32PWM::allocateTimer(3);
	Serial.begin(115200);
	pwm.attachPin(APin, freq, 10); // 1KHz 8 bit

}
void loop() {

	// fade the LED on thisPin from off to brightest:
	for (float brightness = 0; brightness <= 0.5; brightness += 0.001) {
		// Write a unit vector value from 0.0 to 1.0
		pwm.writeScaled(brightness);
		delay(2);
	}
	//delay(1000);
	// fade the LED on thisPin from brithstest to off:
	for (float brightness = 0.5; brightness >= 0; brightness -= 0.001) {
		freq += 10;
		// Adjust the frequency on the fly with a specific brightness
		// Frequency is in herts and duty cycle is a unit vector 0.0 to 1.0
		pwm.adjustFrequency(freq, brightness); // update the time base of the PWM
		delay(2);
	}
	// pause between LEDs:
	delay(1000);
	freq = 1000;
	pwm.adjustFrequency(freq, 0.0);    // reset the time base
}`

Can anyone help me?

Thank You
Michael from Germany

reddy9698, 2022-02-15:
Hi,
The SG90 servos work at a Max frequency of 50Hz. The code that you are using is for LEDs and not suitable for servo motor.

`#include <ESP32Servo.h>
 
Servo myservo;  // create servo object to control a servo
// 16 servo objects can be created on the ESP32
 
int pos = 0;    // variable to store the servo position
//Important :  Recommended PWM GPIO pins on the ESP32 include 2,4,12-19,21-23,25-27,32-33 
int servoPin = 13;
 
void setup() {
	// Allow allocation of all timers
	ESP32PWM::allocateTimer(0);
	ESP32PWM::allocateTimer(1);
	ESP32PWM::allocateTimer(2);
	ESP32PWM::allocateTimer(3);
	myservo.setPeriodHertz(50);                // standard 50 hz servo
	myservo.attach(servoPin, 500, 2400);  // attaches the servo on pin 18 to the servo object
	// using default min/max of 1000us and 2000us
	// different servos may require different min/max settings
	// for an accurate 0 to 180 sweep
}
 
void loop() {
 
	for (pos = 0; pos <= 180; pos += 1) { // goes from 0 degrees to 180 degrees
		// in steps of 1 degree
		myservo.write(pos);    // tell servo to go to position in variable 'pos'
		delay(15);             // waits 15ms for the servo to reach the position
	}
	for (pos = 180; pos >= 0; pos -= 1) { // goes from 180 degrees to 0 degrees
		myservo.write(pos);    // tell servo to go to position in variable 'pos'
		delay(15);             // waits 15ms for the servo to reach the position
	}
}`

You could try this code and let me know if it works.

MichaelLeipzig, 2022-02-18:
Unfortunatly it didn´t work. And so I crashed the next servo :(

reddy9698, 2022-02-18:
Then I guess it has to be something to do with your power supply to the servo. Just make sure the Voltage and Current that you're supplying to the servo are correct. 
And I have a doubt by crashed you mean that they aren't working anymore? Did you try the same servo with another MCU like Arduino just for a test if it's still working?