Xinyuan-LilyGO/LilyGO-T-SIM7000G #163: Constantly disconnects

kedzior-io, 2022-02-20:
I'm connecting to my own mqtt broker and testing the how stable the connection is and I'm getting a lot of modem disconnects.  

I'm wondering of this is something related to the code or the network. I wouldn't say network coverage is a problem as I have tower nearby (also in city center). The device is powered with a proper 5V power supply. 

This my log of when device is getting disconnected: 

```
[19/02/2022 00:25]
Connected

[19/02/2022 00:38]
Connected

[19/02/2022 02:45]
Connected

[19/02/2022 05:49]
Connected

[19/02/2022 06:35]
Connected

[19/02/2022 08:33]
Connected

[19/02/2022 09:06]
Connected

[19/02/2022 09:21]
Connected

[19/02/2022 09:26]
Connected

[19/02/2022 09:28]
Connected

[19/02/2022 09:30]
Connected

[19/02/2022 09:32]
Connected

[19/02/2022 09:37]
Connected

[19/02/2022 09:48]
Connected

[19/02/2022 09:53]
Connected

[19/02/2022 09:55]
Connected

[19/02/2022 10:32]
Connected

[19/02/2022 10:34]
Connected

[19/02/2022 10:43]
Connected

[19/02/2022 10:49]
Connected

[19/02/2022 11:08]
Connected

[... 520 lines cut ...]
      
      digitalWrite(LED_PIN, HIGH); 
      SerialMon.println(F("MQTT is not connected"));   
     
      SerialMon.println(F("!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!"));    

      uint32_t t = millis();
      if (t - lastReconnectAttempt > 5000L) 
      {
        lastReconnectAttempt = t;
        if (mqttConnect()) 
        { 
          lastReconnectAttempt = 0;
          digitalWrite(LED_PIN, LOW); 
        }
      }
      
      delay(100);
      return;
    }

  mqtt.loop();
}

```

emigbur, 2022-02-23:
Im running more or less same config, my mqtt broker is configured over raspberry pi, docker mosquito
I will log the disconnection to check. 
I do the check each minute, I mean this function  "if (!mqtt.connected())"

kedzior-io, 2022-03-08:
@emigbur , have you had a chance to check your logs? 
I'm wondering if this is related to the network provider or maybe there is a memory leak somewhere and it restarts the board.

jupiter8888, 2022-04-02:
`  // Connect to MQTT Broker
  boolean status = mqtt.connect("home");`

By any chance do you have another device with the same MQTT Client name? It has to be unique.

kedzior-io, 2022-04-24:
@jupiter8888 nope. That's the only device connected to my MQTT broker. I think it´s more of a modem issue.

superpance, 2024-08-21:
Hi @kedzior-io 
  Did you solve the issue ? I have similar problem with SIM7000E 

Many thanks in advance.
Kind regards,
Angel

cho0p, 2024-09-05:
I have a similar issue. Connection is not reliable. Sometimes it works for a few hours without a disconnect and sometimes it disconnects and reconnects every few minutes. Power supply and cellular signal are sufficient. I also ordered another antenna but same behaviour. I use the Tinygsm and pubsubclient libraries. In the debug log I can only see:

00:01:24:155 -> DATA ACCEPT:0,31
00:01:24:215 -> [726630] ### Unhandled: DATA ACCEPT:0,31
00:01:24:215 -> AT+CIPCLOSE=0

but I haven't been able to find the rootcause yet.

kedzior-io, 2024-09-07:
@superpance & @cho0p I gave up on the device. Pivoted to Raspberry Pi Zero with USB LTE modem. Way faster to get it running and way easier to code and debug. 🤷