Xinyuan-LilyGO/LilyGO-T-SIM7000G #165: TTGO constantly restarts with the following message: Brownout detector was triggered

Juxn3, 2022-03-09:
We have some TTGO-SIM7000G devices with problems.

The TTGO device has a 2500mAh 18650 battery and a 6V8W solar panel, we have tested by connecting the panel directly and also including a 7805 linear regulator to the TTGO, in both cases the following problem occurs.

When the battery is fully charged (indicator LED green) and the solar panel is exposed to a good amount of sunlight (supplying 5VDC), the TTGO constantly restarts with the following message:

Brownout detector was triggered

et Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clockdiv:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8

This stops happening when the solar panel is disconnected, or when the battery begins to discharge, for this reason we can assure that it is a problem between the battery charge and the solar panel.

tulingBerg, 2022-05-29:
I have the same problem and in addition the Sim module does not get a connection when powered by the battery. Only if powered with USB it connects. Is there a problem with some voltage regulators? Any idea how I can resolve this?
Looking for help ...

LilyGO, 2022-05-30:
> We have some TTGO-SIM7000G devices with problems.
> 
> The TTGO device has a 2500mAh 18650 battery and a 6V8W solar panel, we have tested by connecting the solar panel directly and also including a 7805 linear regulator to the TTGO, in both cases the following problem occurs.
> 
> When the battery is fully charged (indicator LED green) and the solar panel is exposed to a good amount of sunlight (supplying 5VDC), the TTGO constantly restarts with the following message:
> 
> Brownout detector was triggered
> 
> et Jul 29 2019 12:21:46
> 
> rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT) configsip: 0, SPIWP:0xee clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00 mode:DIO, clockdiv:1 load:0x3fff0018,len:4 load:0x3fff001c,len:1044 load:0x40078000,len:10124 load:0x40080400,len:5856 entry 0x400806a8
> 
> This stops happening when the solar panel is disconnected, or when the battery begins to discharge, for this reason we can assure that it is a problem between the battery charge and the solar panel.

You said it works when you disconnect the solar cell?  This could be a problem with the solar panels

LilyGO, 2022-05-30:
> I have the same problem and in addition the Sim module does not get a connection when powered by the battery. Only if powered with USB it connects. Is there a problem with some voltage regulators? Any idea how I can resolve this? Looking for help ...
No connection? Is the status of SIM7000 on?

tulingBerg, 2022-05-31:
Yes, status is on. As I mentioned it works well if powered over USB. Only if powered by the 18650 (fully charged) I have problem with brownout when I switch modem on.

LilyGO, 2022-05-31:
I did not encounter the situation you described when I used 18650.  What 18650 do you use?  How about another one?

tulingBerg, 2022-06-01:
I also tried directly from my DC power supply with same effect. On the vbat at 4.5v it works perfectly (btw what is the min/max rating of vbat?).
Thank you already for caring!

LilyGO, 2022-06-06:
3.5~5v

tulingBerg, 2022-06-06:
Thanks a lot for your support!
So I assume, that the voltage regulator for the 18650 is defect... 😑