Xinyuan-LilyGO/LilyGO-T-SIM7000G #166: TinyGSM Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Raymond-yeo, 2022-03-10:
Hi,
I am new to ESP32. 
I got some scripts to work but when trying to connect to GSM I got this error.
Connecting to Thingspeak and upload my sensor reading works, but when I change script to the tinyGSM All Functions Example, 
I get this error. 

Error keeps on repeating and rebooting
____________________________________________________________________________________________________________________
[6036] Trying baud rate 57600 ...
05:56:28.169 -> Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
05:56:28.215 -> Core 1 register dump:
05:56:28.215 -> PC      : 0x01002822  PS      : 0x00060730  A0      : 0x800d14fe  A1      : 0x3ffb1f20  
05:56:28.215 -> A2      : 0x3ffbfec4  A3      : 0x00002580  A4      : 0x0000e100  A5      : 0x0000e100  
05:56:28.215 -> A6      : 0x00000004  A7      : 0x0000000a  A8      : 0x800d13d4  A9      : 0x3ffb1f00  
05:56:28.215 -> A10     : 0x3ffb1f44  A11     : 0x3ffbfec4  A12     : 0x0800001c  A13     : 0x00000009  
05:56:28.215 -> A14     : 0x00000009  A15     : 0x00000000  SAR     : 0x00000016  EXCCAUSE: 0x00000014  
05:56:28.215 -> EXCVADDR: 0x01002820  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
05:56:28.261 -> 
05:56:28.261 -> ELF file SHA256: 0000000000000000
05:56:28.261 -> 
05:56:28.261 -> Backtrace: 0x01002822:0x3ffb1f20 0x400d14fb:0x3ffb1f80 0x400d452a:0x3ffb1fb0 0x400860ed:0x3ffb1fd0
05:56:28.261 -> 
05:56:28.261 -> Rebooting...
05:56:28.261 -> ets Jul 29 2019 12:21:46
05:56:28.261 -> 
05:56:28.261 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
05:56:28.261 -> configsip: 0, SPIWP:0xee
05:56:28.261 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
05:56:28.261 -> mode:DIO, clock div:1
05:56:28.261 -> load:0x3fff0018,len:4
05:56:28.261 -> load:0x3fff001c,len:1216
05:56:28.261 -> ho 0 tail 12 room 4
05:56:28.261 -> load:0x40078000,len:10944
05:56:28.261 -> load:0x40080400,len:6388
05:56:28.261 -> entry 0x400806b4
05:56:28.399 -> [36] Wait...

________________________________________________________________________________________________________________

LilyGO, 2022-03-11:
Hello, does it work with this example  https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/tree/master/examples/Arduino_TinyGSM/AllFunctions

Raymond-yeo, 2022-03-11:
> [](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/tree/master/examples/Arduino_TinyGSM/AllFunctions)

Hi @LilyGO It is working with this file. 

Thank you very much.  

Can I check with you, if there are any good source for tutorial for this module?

LilyGO, 2022-03-12:
Did you say t-Sim7000G?  There are some examples you can refer to and learn from.