Xinyuan-LilyGO/LilyGO-T-SIM7000G #167: No Network

reddy9698, 2022-03-12:
Hi, I've been using the module for quite some days now and it was working all fine.
Suddenly since yesterday the module doesn't seem to get a network lock. 
I use the same SIM on my phone and it works fine but on the module the status LED keeps blinking every 1s.
The same SIM also works fine on other boards.
I've even tried using other SIM cards but to no avail.
Even tried changing the antenna and still the same.

Is it a problem with the board? Please Help

I even tried the following AT commands : 
AT+CPIN?
=+CPIN : READY

AT+CSQ
=+CSQ : 99,99

AT+CGREG?
+CGREG : 0,2

LilyGO, 2022-03-14:
Hello, how many 7000gs do you want to buy?  How many of them can't connect to the Internet?  Have you configured the network?  Is the antenna properly connected?

reddy9698, 2022-03-14:
I have already bought 5. One module doesn't seem to connect to the network. The Status LED just keeps blinking every 1s. If I use the same SIM card on other boards it is working fine.

LilyGO, 2022-03-15:
+CGREG: 0,2 indicates that the module is in the registered network.  Wait a while and see if you can connect?