Xinyuan-LilyGO/LilyGO-T-SIM7000G #168: Ich suche die Pins für die onboard Led s

tom2323tom2323, 2022-03-16:
Hello, I would like to use the onboard Led s. But I can't find the pin in the datasheet.
Can someone help me

THX

LilyGO, 2022-03-17:
Which LED do you want to know?

mvandermeulen, 2022-03-20:
Se the final page of [SIM7000G_20191227.pdf](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/schematic/SIM7000G_20200415.pdf) does that give you the information you are after?

WolcottSystems, 2022-03-20:
The blue LED should be pin 12 on version 20200415.

realbeeit, 2022-04-01:
To shut down blue power LED you need to set PIN 12 to HIGH! Not LOW!

pitlochry, 2022-06-16:
How to shut down the green (and red) LED? Especially the green LED is bright as hell.

LilyGO, 2022-06-17:
Red is the battery is in progress, green is charging finished.  Unable to close using software.  You can remove it if you don't need it