Xinyuan-LilyGO/LilyGO-T-SIM7000G #170: If disconnected and plugged back to USB-C terminal does not work

Gouzi001, 2022-03-21:
1. you program LilyGO and terminal working fine
2. disconnect LilyGO - still working - doing programmed actions as it should
3. plug it back to USB-C -> terminal does not work 

4. Restart LilyGO  by button still being connected - terminal still does not work
5. Reprogramm it -  Terminal working.

LilyGO, 2022-03-21:
Hello, which example are you using?  Can we make a video to demonstrate that?

Gouzi001, 2022-03-21:
It does not matter what you upload, you can try simple   Serial.print(millis); delay(1000); ...unplug LillyGO plug it back and there will be no results in Terminal. In other words seems that Serial is not being reinitialized.

LilyGO, 2022-03-21:
Calling the millis function is "millis()".   Serial.print(millis());

Gouzi001, 2022-03-21:
not important my friend.. simple disconnect LILIGO plug it back and terminal output will not work anymore... even after hard reset. :(

LilyGO, 2022-03-22:
What printing tool do you use?  Is it normal to use simple printing?  (such as)  
`
uint32_t chipId = 0;
void setup() {
	Serial.begin(115200);
}

void loop() {
	for(int i=0; i<17; i=i+8) {
	  chipId |= ((ESP.getEfuseMac() >> (40 - i)) & 0xff) << i;
	}

	Serial.printf("ESP32 Chip model = %s Rev %d\n", ESP.getChipModel(), ESP.getChipRevision());
	Serial.printf("This chip has %d cores\n", ESP.getChipCores());
  Serial.print("Chip ID: "); Serial.println(chipId);
  
	delay(3000);

}
`

Can you send a video demo that doesn't print properly?

LilyGO, 2022-03-22:
What printing tool do you use? Is it normal to use simple printing? (such as)

//
uint32_t chipId = 0;
void setup() {
Serial.begin(115200);
}

void loop() {
for(int i=0; i<17; i=i+8) {
chipId |= ((ESP.getEfuseMac() >> (40 - i)) & 0xff) << i;
}

Serial.printf("ESP32 Chip model = %s Rev %d\n", ESP.getChipModel(), ESP.getChipRevision());
Serial.printf("This chip has %d cores\n", ESP.getChipCores());
Serial.print("Chip ID: "); Serial.println(chipId);

delay(3000);
}
//

Can you send a video demo that doesn't print properly?

LilyGO, 2022-03-25:
This is most likely an Arduino terminal problem.  You need to restart the terminal after you disconnect the USB.

Gouzi001, 2022-03-25:
i tested to open & close terminal also whole IDE, however with same result as you seen on Video.
For Arduino it's working fine.
Same behavior is on other PC with Linux.

I shall receive this week additional LILIGO and check if they will act better.