Xinyuan-LilyGO/LilyGO-T-SIM7000G #171: Not working connection between ESP and SIM7000

dizider, 2022-03-24:
Hi :wave:,

I was interested in this module so I bought one and it came today. So first I was trying a basic example [ATdebug](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/examples/Arduino_TinyGSM/ATdebug/ATdebug.ino) and unfortunately the code does not work. After that, I tried all other examples with the same result. The ESP32 can not connect to SIM7000G (respectively I am not sure if it is a connection problem but on every command, I got `0x00` as a response or nothing)

If I connect SIM7000 directly (through pins 26 and 27) to the PC everything works great. Is there anybody who solved a similar problem or do have you any idea what to do next? To be true, I am a little bit upset because I bought this module for its simplicity. I hope that I just miss something and the module is not defective. 

Thanks for help

dizider, 2022-03-24:
UPDATE:
I figured out that the problem is between a chair and keyboard, obviously. After a few hours of sadness, I tried a micropython example, and voila, it works. Now I am sure that I missing something important about ESP32 Cpp serial api. Does anyone have any idea? I am confused :confused:

LilyGO, 2022-03-25:
Use [ATdebug](HTTP: / / https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/examples/Arduino_TinyGSM/ATdebug/ATdebug.ino)What does a serial port monitor print?

LilyGO, 2022-03-25:
Use [ATdebug] What does a serial port monitor print?

dizider, 2022-03-25:
Here is a dump of communication: 
```
Testing Modem Response...

****
␀␀␀␀␀␀␀␀␀␀␀␀␀␀
␀␀␀␀␀␀␀␀␀␀␀␀␀
␀␀␀␀␀␀␀␀␀␀␀␀␀
␀␀␀␀␀␀␀␀␀␀␀␀␀
␀␀␀␀␀␀␀␀␀␀␀␀␀
␀␀␀␀␀␀␀␀␀␀␀␀␀
```

dizider, 2022-03-25:
Now I observed something that blew my mind! My previous thought that the problem is somewhere in UART configuration is gone. When I try to find out what's going on in the communication and I find a surprising thing -- communication "works well" when I connect the USB-UART convertor to pins 26 and 27.
```
Starting Up Modem...

aatt

OK
aatt

OK

```

LilyGO, 2022-03-26:
The serial port monitor is abnormal.  Can you take a screenshot?  Or record a video demo?

jendakol, 2022-03-26:
Hi @LilyGO, I'm @dizider's colleague ;-) He's not available right now but I'll try to answer. Unfortunately, I don't have the device so I'm unable to provide you with the screenshot. However, it wouldn't be of any help anyway, the output is just what is written above.

Maybe let me recapitulate what we have tried:
- Running C++ examples, provided in the repo, doesn't work. The modem is never connected, the output is _weird_.
- Surprisingly, when running microPython examples, everything is smooth and correct.
- When talking to the SIM module directly (with USB<->UART converter connected to pins 26/27), the SIM module works as expected.
- When the USB<->UART converted is connected to the SIM module pins, even the C++ code works. The converter doesn't have to be connected to the PC nor to Vcc or GND of the board - just the two pins.

So... do you have *any* idea what could cause this weird behavior?
- Could it be some HW issue?
- Could it be wrong Platformio setup (e.g. wrong board selected? We tried `esp-wrover-kit` and `esp32dev`)
- anything else?

We are really desperate :-) Thx.

dizider, 2022-03-27:
For you, who are solving a similar problem is here a solution: **use a newer version of Arduino ESP32** framework. If are you using Platformio, here is an example of `platformio.ini` file or you can check our project ([GPS tracker](https://github.com/litvinov-tabor2022/lightweight-gps-tracker)) where we are using this module :slightly_smiling_face: 

```
[env:esp32dev]
platform = https://github.com/Jason2866/platform-espressif32.git
board = esp32dev
framework = arduino
platform_packages =
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
```

closed 2022-03-27 by dizider