Xinyuan-LilyGO/LilyGO-T-SIM7000G #173: LTE CAT-M issue in Switzerland

vladco78, 2022-04-16:
Hello Lilygo, I have about 20 T-SIM7000G in use in Switzerland, now 2G has been completely switched off and since then all boards have been offline.  The SIM cards are from 1NCE with LTE CAT-M support.  What could be the problem?  Can anyone confirm this board works over LTE CAT-M in Switzerland?

LilyGO, 2022-04-18:
This is the band supported by 7000 GIGABytes.  Make sure your SIM card is in one of these bands
[image]

num-lock, 2022-04-19:
@vladco78 I have few dozen boards online with cards from 1nce and some with Hologram as well. Four of them are in Switzerland without any issues. Have you activated LTE on the modem itself? I can confirm LTE Cat.M1 and NBIoT are working fine. YMMV depending on your cellular coverage.

You might have to set it explicit, e.g. with `modem.setPreferredMode(38);` for LTE only or `2` instead of `38` for automatic mode which should cover all available technologies if you didn't do that already.

vladco78, 2022-04-19:
Yes, I switched to lte-m manually AT+CNMP=38, the modem is trying to connect all the time, I can see that in the 1NCE logs.  According to 1NCE, only CAT-M works in Switzerland!  Do you use the original enclosed full-band lte antenna?

vladco78, 2022-04-21:
Update, 3 devices in Berlin work via LTE CAT-M + 1NCE SIM. In many other places I get no connection neither LTE CAT-M nor NB-IoT.

LilyGO, 2022-04-22:
You test using [examples ](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/tree/master/examples/Arduino_TinyGSM/ATdebug.) .
Send AT instructions in turn:
1.  AT+CFUN=0
2.  AT+CNMP=2
3.  AT+CFUN=1
4.  AT+CPSI? 
It takes some time to register the network. When the registration is successful, send AT+CPSI?  It will display information.

vladco78, 2022-04-26:
Update!
I've done some testing now, if it doesn't work then it's probably the provider. Sunrise had problems before, now 4G LTE CAT-M works great! G2 will also be operated via Sunrise until the end of 2022! 
Thanks to all!

closed 2022-04-26 by vladco78