Xinyuan-LilyGO/LilyGO-T-SIM7000G #176: SIM7000 dead?

MalteHillmann, 2022-05-02:
Hello!

I have purchased a board from AliExpress (LilyGo-Store) and everything worked fine.
But last week ago, the board stopped reporting values. 
A short investigation shows: The SIM7000G does not respond to any commands. There is no answer on the serial port at all. The two LEDs near the SIM7000 (NETLIGHT/STATUS) are off, too. (No continuous/blinking)
Is the SIM7000G dead or is there a chance to recover that board?

Thanks in advance.
Kind regards

LilyGO, 2022-05-03:
Hello,Can you see if this [example](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/tree/master/examples/Arduino_TinyGSM/AllFunctions) works?  View the information printed by the serial port monitor?

MalteHillmann, 2022-05-24:
I've tested the example already. The modem doesn't even reply to an "AT". It replies nothing. 
So the example only shows:
`Failed to restart modem, attempting to continue without restarting`

Is there anything else I can do?

LilyGO, 2022-05-25:
Whether the status indicator (red) of 7000G is on？

MalteHillmann, 2022-05-27:
Neither of the two LEDs from the SIM7000 are lit.

LilyGO, 2022-05-28:
[reference](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/docs/How%z%20update%20firmware.md).
Connect to 5V DM DP GND is there a COM port?

MalteHillmann, 2022-05-30:
I've done that but other than expected I don't get an "SimTech SIM7000" device under "Other devices". But I get an "QHUSB_BULK"-Device (VID 05C6, PID 9008) which seems to be an Qualcomm Bootloader. I've tried the driver installation and got "SimTech HS-USB QDLoader 9008" (COM Port) Device. Then I tried to flash the Modem as shown in the upgrade guide.
Sadly the update stopped at some point:
[image]
I tried to repower the modem as stated in the error, but that doesn't help. Again the same error.

Now the Modem seems to be stuck in factory test mode (because IMEI is missing?): 
```
-> AT+CGMI
<- SIMCOM_Ltd
<- OK
-> AT+CGMM
<- SIMCOM_SIM7000A
<- OK
-> AT+CGSN
<- +CME ERROR: memory failure
-> AT+CFUN?
<- +CFUN: 5
<- OK
-> AT+CFUN=1,1
<- OK
<- RDY
<- +CFUN: 5
<- +CPIN: READY
<- SMS Ready 
```

Is there a way to repair that?

Mr-HaleYa, 2022-07-29:
They do die sometimes, especially if you were using the battery power. I've bought about 120 of them and about 20 have died or I killed them accidentally... I found that if the battery dies and then gets charged via solar it has a high chance of dying. seems to be something to do with a lack of proper hysteresis on the chip that powers the ESP 32 causing _Rapid_ power cycling that fries it. I've also had the host chip die and even a SIM7000G die. 

Stuff dies... Buy another and try again is about all you can do, if it happens again then you're probably doing _somthing_ wrong.

MalteHillmann, 2022-09-21:
> Maybe you've configured some sleep commands?
No, never changed anything like this in the SIM7000.

After that reflash on 30.05. it works great.

But now I have another problem with **exactly the same board**. Software on it works, SIM7000G works. 
When I connect USB, it shuts down everything instantly (all LEDs off) and shows red charging LED only. 
If I plug it to a PC it does the same, but recognizes the CP210x correctly, but I'm not able to flash a new firmware anymore, because the ESP32 can't be reached. (not powered at all?)

Any ideas?

Thanks in advance.
Kind regards