Xinyuan-LilyGO/LilyGO-T-SIM7000G #177: SIM7070G unable to find a GNSS fix

sw-dev-code, 2022-05-03:
I have a SIM7070G module with which I’m trying to get a GNSS fix but without success. Below you can see my sequence of AT commands and responses.

```
AT+CGNSPWR=1
OK

AT+CGNSINF
+CGNSINF: 1,,,44.060004,20.920002,-37.592,,,1,,0.1,0.1,0.1,,,,155976.0,6000.0
OK

AT+CGNSINF
+CGNSINF: 1,,,44.060004,20.920002,-37.592,,,1,,0.1,0.1,0.1,,,,155976.0,6000.0
OK

AT+CGNSINF
+CGNSINF: 1,,,44.060004,20.920002,-37.592,,,1,,0.1,0.1,0.1,,,,155976.0,6000.0
OK

AT+CGNSINF
+CGNSINF: 1,,,44.060004,20.920002,-37.592,,,1,,0.1,0.1,0.1,,,,155976.0,6000.0
OK

…
```

I keep getting the same response with always the same latitude and longitude which aren’t correct. Can you tell me what can be wrong?

Thank you in advance.

LilyGO, 2022-05-04:
Hello，What is your correct latitude and longitude ? Is it very different from the latitude and longitude you ge t?

sw-dev-code, 2022-05-04:
It's very different, my real location is about 150 km away from the reported one. And other parameters are also very strange, like timestamp and number of satellites.

LilyGO, 2022-05-04:
Judging by the results of your GPS.  Time has not been obtained, the information obtained is not complete.  You need to keep positioning.  Did you get a location outside?  Sometimes it's affected by the weather.  Slow positioning .

sw-dev-code, 2022-05-04:
I'm testing it outside with a clear weather and open sky. Still getting the same results.

Below you can see a picture of my setup.

[image]

LilyGO, 2022-05-05:
You may have fewer satellites in your area.  You need to stay outside and wait a little longer .

AT+CGNSINF
+CGNSINF: 1,,,44.060004,20.920002,-37.592,,,1,,0.1,0.1,0.1,,,,155976.0,6000.0
OK

The results show that there is data, but it is not complete.

Mr-HaleYa, 2022-07-29:
You need to go outside where there are few trees. A roof works really well, try it there. Also, it can take up to 5 min to get the first location. Do that then report your results.

xyzzy42, 2023-04-14:
It should work inside just fine if near a window.  Avoiding trees or being a on a roof is not necessary.

@sw-dev-code, your problem may be that the SIM7070G GPS antenna power is on a different GPIO that the documentation states, as it only covers SIM7000G.

Try this command:
```
AT+SGPIO=0,5,1,1
```

Then you should be able measure ~3V on the pad of the GPS connector and top pad of the capacitor directly below it.

OrnellaBenzi, 2023-06-26:
i'm having a similar problem, when using AT+CGNSINF i get inaccurate and incomplete data. And when using AT+SGNSCMD I keep getting +SGNSERR: 10 after a minute or two:
[image]

albertmarc8, 2025-02-24:
@OrnellaBenzi have you been able to solve this issue or know the cause? I'm trying on a SIM7080G devboard and I get the same SGNSERR.