Xinyuan-LilyGO/LilyGO-T-SIM7000G #181: sending data to google spreadsheet

Fishtronics, 2022-06-14:
Hi am trying to send data to google spreadsheet following the tutorial
https://www.electroniclinic.com/google-spreadsheet-or-google-sheets-with-esp8266-nodemcu-for-data-logging/
but the example shows only how to do it via wifi not via a sim card.
I Have tested the lilygo see link:
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/examples/Arduino_TinyGSM/AllFunctions/AllFunctions.ino
and
https://github.com/botletics/SIM7000-LTE-Shield/blob/master/Code/examples/LTE_Demo/LTE_Demo.ino
works fine but I don't know how to get it to work with the google spreadsheet?
Can you please help me?
Thanks

icabrera29, 2023-08-17:
Same problem here :/

Maximumotion, 2024-04-18:
Just came across this topic as well. Was anyone able to do it?

zacsto, 2024-04-18:
Yea I did, I believe the issue was sending over secured protocol. I ran in to other issues with the board randomly hanging during connecting and needing to be physically reset so haven't worked on it in months.

akm2025, 2024-10-07:
hola alguien ha logrado enviar datos desde lilygo esp32sim7000g a google sheets usnado datos de internet del simcard?