Xinyuan-LilyGO/LilyGO-T-SIM7000G #186: Inconsistent Power On Modem code in examples

DevinCarpenter, 2022-07-04:
Hi guys, I'm seeing inconsistent examples of powering on the modem from what's located inside the examples folder. I don't have issues powering the modem on - but I'm concerned that not powering on and rebooting the modem correctly will lead to issues and may be causing some current issues I am having.

In the examples folder projects, the ways that are shown to power on the modem with pin 4 are as follows:
```
lobo examples            - high, wait 300ms, low
cayenne example          - high, wait 300ms, low
arduino_gps              - low, wait 1000ms, high
Arduino_NetworkTest.ino  - low, wait 1000ms, high
AllFunctions.ino         - high, wait 300ms, low
ATdebug.ino              - high, wait 300ms, low
ResetModem.ino           - high, wait 300ms, low
Blynk_Console.ino        - high, wait 300ms, low
MqttClient.ino           - high, wait 300ms, low
Platformio example       - high, wait 10ms, low, wait 1010ms, high
```

which way is the proper method?

I'm not sure where the 300ms even comes from, since as per the Sim7000 hardware design doc the powerkey input starts low with the modem disconnected from battery, goes high when connected to battery for an indeterminate time, goes low for a minimum of 1 second, then goes high again - which the platformio example seems to do best - with the arduino gps and network test examples coming in close second.

LilyGO, 2022-07-05:
Hello, You can forcibly shut down SIM7000 in Setup () and then boot it up
：
  pinMode(MODEM_PWKEY, OUTPUT);
  
  //SIM7000 OFF
  digitalWrite(MODEM_PWKEY, HIGH);
  delay(6000);//Ensure that SIM7000 is powered off after a long delay

  //SIM7000 ON
  digitalWrite(MODEM_PWKEY, LOW);
  delay(300);
  digitalWrite(MODEM_PWKEY, HIGH);
   delay(2000);
  digitalWrite(MODEM_PWKEY, LOW);
  delay(300);

Mr-HaleYa, 2022-07-29:
I wrote the code in the ones that are 300 ms, and it works. You can increase it if you want but it's not needed from my testing. Honestly didn't look into it too much, I kinda just chose a number and it worked so I stuck with it lol. 
My code in my personal sketch is this
```c 
  pinMode(MODEM_PWKEY, OUTPUT);                     // Set-up modem  power pin
  digitalWrite(MODEM_PWKEY, HIGH);
  delay(300);
  digitalWrite(MODEM_PWKEY, LOW);
  delay(6000);
```

This code is running on about 58 devices and works perfectly.

---

So what's your actual problem?

DevinCarpenter, 2022-07-29:
@LilyGO Thanks for the reply. Your power on code does work for me. However the power off code does not make any sense - as pin 4 is set low when powering the modem on - so setting it to low again will not do anything. Your Power On code also works as a "toggle" power function rather than just a power on - meaning I can turn the modem on or off with this code.

@Mr-HaleYa Appreciate you answering. I don't have a problem as much as I was just wondering why the example sketches were so inconsistent in how the modem was powered on. It would seem that leaving Pin 4 high or low and the timing in-between shouldn't be something that is arbitrary as in the examples.

Overall I was just worried that issues might crop up if done incorrectly with the timing and leaving pin high when it should be low - or vice versa.

Currently my code for turning the modem on is working - but it is more of a power "toggle" as stated above with LilyGo's code. I was trying to find the correct approach using the SIMCOM hardware guidelines but I couldn't get what the document was saying to work.

Using Micropython my code has been pretty close to LilyGo's:

    MODEM_PWKEY.value(0)
    time.sleep_ms(300)
    MODEM_PWKEY.value(1)
    time.sleep_ms(2000)
    MODEM_PWKEY.value(0)
    time.sleep_ms(15000) #arbitrary time to wait for modem to power on properly

I just tested your code though - and it does work well for me as just a power on function instead of a toggle, so that's good. Also glad to hear it's working reliably for you as well on so many modules.

You wouldn't happen to have working code for a power off or reset via hardware pins?

Mr-HaleYa, 2022-07-29:
this is my reboot code, it's not Micropython but you should get the general idea
```c
void modem_reset() {
  Serial.println("\nModem hardware reset\n");
  gprsStatus = modem.isGprsConnected();   // Get the gprs status, if connected to network or not
  if (gprsStatus != 0) {
    modem.gprsDisconnect();    // Disconnect if connected, makes it show as a clean disconnect instead of unexpected
    Serial.println("GPRS Disconnected");
  }
  if (!modem.restart()) {  // Try using the built-in restart function, this will fail if the module is frozen or busy
    Serial.println("Failed to restart modem!");
    pinMode(MODEM_RST, OUTPUT);
    digitalWrite(MODEM_RST, LOW);
    delay(260);  // Time to hold for reset is 252ms
    digitalWrite(MODEM_RST, HIGH);
    delay(15000);       // Modem takes longer to get ready and reply after this kind of reset vs power on
  } else {
    Serial.println("Modem Restarted");
    delay(10000);  // wait for modem to fully boot up before resuming code or you will get really weird behavior
  }
}
```

DevinCarpenter, 2022-07-29:
Great, I'll test the code on my side too. What is the reset pin? I think I saw on another post that you were using pin 5 - but I'm not sure how you derived that pin. I've just been going on the pin table from the main product page.

Mr-HaleYa, 2022-07-29:
```c
#define MODEM_PWKEY     4   /* sim7000G power pin */
#define MODEM_RST       5   /* sim7000G reset pin */
```

vkcorp, 2022-11-23:
I'm also wondering why each same is different to turn on/off the modem. 
It's toggling with how much <ms>... Thanks too !

May I ask one more ? I found MODEM_PWKEY or PIN_DTR usually defined as 25. 
Does it has to be set as 25 ? Or can I change it with other pin number ?
If I can change it is there any condition or check-point to consider ?
Actually, I guess it's GPIO number and I will use the GPIO 25 to another purpose..

DevinCarpenter, 2022-12-03:
@vkcorp I'm currently getting by with ms times close to Mr Hale Ya for powering/resetting/toggling. 

As for the PIN_DTR...I believe you can use Pin 25 as a standard GPIO pin as it shows in the diagram. As far as re-assigning another pin to be PIN_DTR - I don't know. I don't use PIN_DTR and I haven't reassigned any pin functionalities by moving them to other pins. I either used the pins for their intended purpose,  assign them for sensor communication, or I don't use them.

vkcorp, 2022-12-05:
I see and thanks for you advice, @DevinCarpenter  !!