Xinyuan-LilyGO/LilyGO-T-SIM7000G #189: my win7 puter can't find the port

jimbooth97, 2022-08-27:
Just got this lilygo sim7000 and tried to connect to the puter, but can't find the port.  board does light up and my puter makes the sound that i just connected, but in the arduino ide, i dont get a port number.  I do have a battery connected. Any suggestions?

jimbooth97, 2022-08-27:
just tried a good idea at this link: https://github.com/botletics/SIM7000-LTE-Shield/wiki/USB-Drivers  but didn't work.  Was pretty surprised. However it suggested that i would see the sim7000 in the PORTs but all i saw, was under "Other Devices/USB Single Serial" with a caution mark. That is the lilygo board, and i tried to install from the win7 unzipped folder but no luck.  I have a win 10 puter, will try that later.

DevinCarpenter, 2023-02-08:
When you're connecting the USB to your computer - you're not connecting the SIM7000 - you're connecting the ESP32.

Botletics boards don't have an ESP32, so that solution will not work for you.

You will want to install the necessary libraries to interface with the ESP32 using the Arduino IDE.

I would recommend following [this tutorial](https://randomnerdtutorials.com/installing-the-esp32-board-in-arduino-ide-windows-instructions/).