Xinyuan-LilyGO/LilyGO-T-SIM7000G #191: Cellular in Canada?

aneeshraghu, 2022-09-04:
Hey guys,

Was anyone able to get this work with cellular in Canada? If so, how? and on which network?

I have a hard time getting it to work with Bell or Telus. When I run the network test all I get is not connected.

neostim, 2022-12-05:
For what it's worth I was able to get it working on Telus in Canada, it took some effort, I ended up using the "Diagnostics" example and got it consistently connected from there.

aneeshraghu, 2022-12-05:
That didn't work for me. I ended up having to use truphone.

neostim, 2022-12-06:
Funny enough I just tried setting up my 2nd one on Telus and it's not connecting, Keeps saying it failed to connect or failed to initialize modem, the same SIM card works fine in my first one .... so either one of the many attempts I made early on set some AT command correctly on my first modem or this second device is just bad...

aneeshraghu, 2022-12-06:
That sucks man. I have a bunch of the boards none of them worked with the major carriers mainly because they no longer support 2g or 3g.

Sell24, 2023-01-26:
Take a look at https://www.hologram.io/ They support 3G and their prices are decent.

neostim, 2023-01-26:
> That sucks man. I have a bunch of the boards none of them worked with the major carriers mainly because they no longer support 2g or 3g.

So I have 10 of these now and I had the same issue on a couple and managed to get them and my original one fixed, it’s a combination of running their diagnostics sketch, then running my normal sketch then resetting it when it fails to connect, it’s flawlessly worked 4 times now and they have continued to work all month, I’m assuming it’s some combination of AT commands being sent. 

I’m on mobile now but can give more detail later if interested

jrkswk, 2023-02-07:
neotim, I'd be interested in the combination you used. I'm in the USA but having similar issues with two different compnay's SIM cards.

virgilHirtz, 2023-10-14:
@neostim Do you have more information on how you got your boards set up? I'm struggling to get mine connected properly. Where did you find the diagnostics sketch?

neostim, 2023-10-14:
I’m not back at work until Monday but can send over more details then, the
diagnostics sketch was on lilygo’s github if I remember correctly.

Tom-E, 2023-10-18:
Same as well, also stuck trying to get connected to Telus IoT and any help would be appreciated

virgilHirtz, 2023-10-18:
@Tom-E I managed to get mine functioning after running the diagnostics sketch buried in lilygo's git page, after running that all my functions worked, and tested well using the All functions example with the tinygsm library. I also noticed on some examples I previously attempted to run the pinout to initialize the modem was wrong and needed to adjust that before it worked.