Xinyuan-LilyGO/LilyGO-T-SIM7000G #196: Sim7000g without Arduino (ESP-IDF)

dgomlopez, 2022-10-11:
Hello, 

do you know if it is possible to somehow make the sim7000 (gsm + gps connection) work correctly with the esp-idf?

With Arduino it seems to work fine, but trying to replicate that functionality in the ESP-IDF is getting quite complicated.

I am trying to use esp_modem as a component of ESP-IDF.

Any help is welcome.

Thank you!

pixelskull, 2022-10-24:
Trying to do the same. Adding esp_modem was possible by adding `list(APPEND EXTRA_COMPONENT_DIRS esp_modem)` within the toplevel **CMakeLists.txt**. For now I am stuck with the pppos_client example, which is not using the gprs features. Did you (@dgomlopez) found a way to solve your problem?

dgomlopez, 2022-10-28:
Hi @pixelskull . I can use esp_modem as a component, but I have not been able to connect to the network or use the gps. esp_modem in principle works with sim800 and sim7600 (I don't know if completely, network + gps), and it seems that it has sim7000 components but it doesn't finish working.

pixelskull, 2022-10-28:
@dgomlopez Same here, I run into error code 263 - which seems like a timeout. Sadly I can't help you, but maybe we can get an answer together through this issue. You are using the pppos_client.c example as well?

dgomlopez, 2022-10-30:
Hi @pixelskull I try to use now this example: https://github.com/espressif/esp-protocols/tree/master/components/esp_modem/examples/simple_cmux_client
 esp_modem 0.1.24 (last version). This example include sim7000 support but dont work apparently.
 
I'm try to use ppos-client.c too, but dont work for the moment.

libertybell95, 2024-12-20:
Bump. Anyone have any luck? I've been having trouble getting the device to communicate over UART to send AT commands using ESPIDF. It works fine when using the TinyGSM library on Arduino.