Xinyuan-LilyGO/LilyGO-T-SIM7000G #201: Modem cannot connect under 3.7v

liorisakov, 2022-12-05:
Hi,
Is it possible that when battery voltage drops below 3.7v the SIM7000g modem cannot connect to network?

In other posts I have seen lowest voltage around 2.7v.

How do you overcome this issue? I dont think thia is relayed to current since its a high quality battery (Samsung E35).

This issue repeats in all my working devices at same voltage level.

vlszabolcs, 2023-01-13:
Which network do you use ?
When I used the GSM network the board often rebooted or freezed because the sim7000 can use 2A . That happens typically far from the  city. I switched to LTE CAT M1 (peak current 0,6 A) and the problem almost ceased. It comes up when the battery is near run low.
You can find more information about the problem in the SIM7000 Hardware Design manual.
I setted up the network via AT-command with [this tutorial](https://m2msupport.net/m2msupport/how-to-use-test-lte-cat-m-with-sim7000-modules/).

DevinCarpenter, 2023-02-04:
I've experienced this same issue. I'm currently connecting to LTE CAT M1 and the modem will not connect at anything lower than 3.7 volts on the battery. Unfortunately I have made this my 0% battery level because of this and would like to solve this as well. I'm testing in the city near a cell tower so I shouldn't be having issues with high power consumption.

After reading the Hardware Design manual v1.07 it states that max voltage is 4.3v and under 2.9v the modem will power off, though it also says VBAT needs a minimum input voltage of 3v.

I also see that using GPRS/EDGE  - the modem will use a maximum of 2A, whereas LTE uses a max of 0.6A - but I didn't see anything else about it.

@vlszabolcs Do you remember where you saw additional information about the problem in the hardware design manual?

fgnievinski, 2023-06-06:
related information:
* https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/227#issuecomment-1508438887