Xinyuan-LilyGO/LilyGO-T-SIM7000G #202: Micro Python example not working

MHfr1, 2022-12-13:
The Micro Python example does not work. When running the code it continuously returns:

[PPPOS CLIENT]: AT BAD RESPONSE: [..+CREG: 0,5....OK..]

I have tested the device using Arduino which does work to eliminate any issues with the board or sim card.

DevinCarpenter, 2023-02-04:
I use Micropython heavily for this board.

What isn't working? You can communicate with the modem using by sending the 'AT' command?

If you're receiving OK from a command then you are definitely establishing communication.

I don't use any of the LilyGo examples as they are not great. The micropython LOBO library is super outdated and you shouldn't be using it either. I've tried them and they didn't work.

I wouldn't use any of the libraries LilyGo has suggested. I use the SPIRAM firmware directly from the micropython website [here](https://micropython.org/download/esp32spiram/).

Did you make sure to download the correct SPIRAM firmware from the Micropython website?

Please provide details and be specific.

MHfr1, 2023-02-04:
Thanks for responding @DevinCarpenter. We were using LOBO as that was what was recommended with the examples so maybe that is where the issues arose.

My understanding was the LOBO was recommended as it included the TinyGSM module but I may be incorrect.

I will give it another go without using their examples.

What we are ultimately trying to achieve is getting GPS coordinates, sync GPS time and ideally to use a http get request to return JSON from our website.

DevinCarpenter, 2023-02-05:
No problem.

I've built my own library as I chose to use python instead of tinygsm because I was facing issues and having to write a lot of C++ to fix them - not my forte.

Getting GPS is fairly easy and once you get it working it works great (within 5-50ft), but it has some quirks.
1. Set the GPS accuracy to be 10 meters. It will try to get best as possible accuracy but it won't ever be farther than 10 meters. I've found that this is the Goldilocks number for me. Use this AT command: AT+CGNSHOR=10
2. Set GPIO 4 (of the modem, NOT the esp32) to be high with command: AT+SGPIO=0,4,1,1
3. Turn on the GPS (we're actually using GNSS, which GPS is a part of) with command: AT+CGNSPWR=1
4. Get data with this command: AT+CGNSINF
5. Repeat until you get data.
6. Set GPIO pin 4 low with: +SGPIO=0,4,1,0
7. Turn off GNSS with: AT+CGNSPWR=0

See [my other post](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/212#issuecomment-1416681132) for additional details on GPS.

FYI, HTTP is fairly easy to get, but not HTTPS. I'm assuming you want the data transfer to be secure.

I use MQTTS. I've gotten HTTP, MQTT, MQTTS, TCP, TCP/TLS all working but HTTPS stumped me. I saw that [some guys at Botletics](https://github.com/botletics/SIM7000-LTE-Shield/issues/156#issuecomment-1332732472) got it working so you can give that a shot once you reach that point.

Good luck guys.

zachelliottwx, 2023-04-06:
Did you ever figure this out? I'm stuck on this as well where the LEDs are on and the modem appears to be on, but I cannot get any AT commands through or anything using LoBo and the gsm module

MHfr1, 2023-04-06:
No, I managed to get GPS working fine but got nothing from GSM. It seemed fine in Arduino but I couldn't get anywhere in microphone.

It seems that Lobo is massively outdated and standard micropython doesn't contain the tinygsm module that Lobo does.

zachelliottwx, 2023-04-06:
Can you share your GPS Code on what you got working?

MHfr1, 2023-04-06:
It was in the micropython example supplied on the manufacturer git.

zachelliottwx, 2023-04-06:
Ok. I can't get that one to work :( Seems like it's having issues with or without a SIM I get this error:
`
I (252018) [PPPOS CLIENT]: Skip command: [AT..]
I (252018) [PPPOS CLIENT]: Skip command: []
I (252018) [PPPOS CLIENT]: Skip command: []
I (252021) [PPPOS CLIENT]: Skip command: []
I (252126) [PPPOS CLIENT]: AT COMMAND: [AT+CPIN?..]
I (252151) [PPPOS CLIENT]: AT BAD RESPONSE: [..ERROR..]
W (252151) [PPPOS CLIENT]: Wrong response, restarting...
I (255151) [PPPOS CLIENT]: Skip command: [AT..]
I (255151) [PPPOS CLIENT]: Skip command: []
I (255151) [PPPOS CLIENT]: Skip command: []
I (255154) [PPPOS CLIENT]: Skip command: []
I (255259) [PPPOS CLIENT]: AT COMMAND: [AT+CPIN?..]
I (255283) [PPPOS CLIENT]: AT BAD RESPONSE: [..ERROR..]
W (255283) [PPPOS CLIENT]: Wrong response, restarting...
.(98, 'Not started')
`

MHfr1, 2023-04-06:
Will send you what I have when I'm back at work